avrdudes/avr-libc #27: [bug #3485] Using float arithmetic without linking with -lm result in incorrect code

avrs-admin, 2022-01-30:
Wed 07 May 2003 05:46:31 PM CEST

Reduced severity as it's a documentation change.

This issue was migrated from https://savannah.nongnu.org/bugs/?3485

avrs-admin, 2022-01-30:
Joerg Wunsch <joerg_wunsch>
Wed 22 Dec 2004 11:11:09 PM CET

We ought to merge libc and libm completely.  There are
other discrepancies as well, like FP conversion functions
listed in <stdlib.h>, while their implementation is only
present in libm.a.

This item will be deferred until after the release of 1.2.0.

avrs-admin, 2022-01-30:
Joerg Wunsch <joerg_wunsch>
Wed 22 Dec 2004 11:35:07 PM CET

Oops, closing this item was **not** intented.

avrs-admin, 2022-01-30:
Joerg Wunsch <joerg_wunsch>
Wed 01 Nov 2006 10:02:30 AM CET

**[Bug #18169](https://savannah.nongnu.org/bugs/?18169)** has been marked as a duplicate.

avrs-admin, 2022-01-30:
Joerg Wunsch <joerg_wunsch>
Thu 20 Mar 2008 06:46:28 AM CET

Well, let's keep this one open until all of our libgcc routines
have really been integrated into libgcc (__fixunssfsi etc.).
 **This** is the actual reason for the incorrect code, and I've
already talked to Dmitry, he wants to look into what needs to
be done to contribute that back to the GCC project.

Unless any of the functions from <math.h> is needed, linking
against libm.a ought to be entirely optional, and should not
cause different code to be generated.

avrs-admin, 2022-01-30:
Georg-Johann Lay <gjlayde>
Sun 02 Sep 2012 09:45:32 PM CEST

See also the discussion in [avr-gcc: Add --with-avrlibc configure option for AVR-Libc](https://savannah.nongnu.orghttp://lists.gnu.org/archive/html/avr-gcc-list/2012-08/msg00070.html)

sprintersb, 2024-02-17:
This was fixed as https://gcc.gnu.org/PR54461 to [GCC v4.7](https://gcc.gnu.org/gcc-4.7/changes.html#avr)
* Remove `float` functions from libgcc that are provided by libm.
* The compiler driver links with `-lm` because libm contains code that the compiler expects in libgcc

closed 2024-02-17 by sprintersb