avrdudes/avr-libc #275: [bug #21805] boot.h: Use out instead of sts

avrs-admin, 2022-01-30:
Sat 15 Dec 2007 12:51:52 AM CET

From boot.h:
\todo From email with Marek: On smaller devices (all except ATmega64/128),
__SPM_REG is in the I/O space, accessible with the shorter "in" and "out"
instructions - since the boot loader has a limited size, this could be an
important optimization.

I'm writing a boot-loader that has to fit into the 256 byte boot-loader space of the 8 kB parts. Here is my proposed solution to fix this matter. I've only fixed the __boot_xxx_normal variants, but the same logic could easily be extended to the __boot_xxx_alternate and __boot_xxx_extended variants.

Cheers,
Shaun

2007-12-14  Shaun Jackman  <sjackman@gmail.com>

- include/avr/boot.h (__boot_page_fill_normal,__boot_page_erase_normal, __boot_page_write_normal,
__boot_rww_enable): Use an out instruction rather than an sts
instruction when possible.

[file #14628: boot.diff](https://savannah.nongnu.org/bugs/download.php?file_id=14628)

This issue was migrated from https://savannah.nongnu.org/bugs/?21805

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Tue 18 Dec 2007 11:48:12 PM CET

Deferred until 1.6.1.

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Wed 19 Mar 2008 11:52:35 PM CET

This is a big enough change that it should be checked on multiple devices. I'm deferring this one again until 1.6.3.

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Tue 30 Sep 2008 04:04:08 PM CEST

Again, deferred until 1.6.4.

avrs-admin, 2022-01-30:
Joerg Wunsch <joerg_wunsch>
Wed 07 Jan 2009 03:56:15 PM CET

Another possible optimization I'm just noticing: the generated
code looks like:

ldi     r24, 0xFF
ldi     r25, 0x03
movw    r30, r24

i.e. a parameter is passed in r24:25, and then moved into
r30:31.  This might IMHO become more optimal (with the
compiler using r30:31 directly) by using the "z" constraint.

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Thu 12 May 2011 09:14:00 PM CEST

I did a test with the original patch, and I don't see how it is saving code space with e.g. the new definition of __boot_page_fill_normal(). I'm using the latest code in boot.h, patched with the patch here (and modified) and gcc 4.3.3. I get 18 bytes of code with the current code and 20 bytes with the new code.

sprintersb, 2024-04-13:
Fixed by https://github.com/avrdudes/avr-libc/commit/ff2e447803fb9867c46d44165363829564bb6a25 and https://github.com/avrdudes/avr-libc/commit/fc349d727f6fe14a40f56e8512f3c0a48d420191.

closed 2024-04-13 by sprintersb