avrdudes/avr-libc #348: [bug #24733] UART0 interrupt vectors definition in iomxx4.h

avrs-admin, 2022-01-30:
Mon 03 Nov 2008 08:37:15 AM CET

Found discrepancy between documentation (online, currently ver 1.6.1)and iomxx4.h (CVS: 1.13.2.3; 1.17) - differences marked with '*'

doc/avr-libc/avr-libc-user-manual/group__avr__interrupts.html  defines

USART0_RX_vect     SIG_*UART0*_RECV     USART0, Rx Complete
USART0_TX_vect     SIG_*UART0*_TRANS     USART0, Tx Complete
USART0_UDRE_vect     SIG_*UART0*_DATA     USART0 Data Register Empty

While iom44x.h defines:

/* USART0, Rx Complete */
#define USART0_RX_vect            _VECTOR(20)
#define SIG_*USART*_RECV            _VECTOR(20)

/* USART0 Data register Empty */
#define USART0_UDRE_vect        _VECTOR(21)
#define SIG_*USART*_DATA            _VECTOR(21)

/* USART0, Tx Complete */
#define USART0_TX_vect            _VECTOR(22)
#define SIG_*USART*_TRANS            _VECTOR(22)

This issue was migrated from https://savannah.nongnu.org/bugs/?24733

avrs-admin, 2022-01-30:
Joerg Wunsch <joerg_wunsch>
Mon 03 Nov 2008 08:26:03 PM CET

First investigation makes me think this is related to the
script that builds vectortable.dox.

It's a documentation-only bug, i. e. the vector names itself
are consistent with the way they have always been in avr-libc
for these devices.  Old-style SIG_XXX vector naming is
deprecated anyway.

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Tue 22 Feb 2011 10:24:04 PM CET

Assigning to Joerg as he is familiar with the patch-headers.py script.

sprintersb, 2025-04-05:
`vectortable.dox` is now auto-generated from the device headers.

closed 2025-04-05 by sprintersb