avrdudes/avr-libc #381: [bug #26734] Unsafe SP register changes in ISR_NOBLOCK interrupt handlers

avrs-admin, 2022-01-30:
Wed 03 Jun 2009 04:10:16 AM CEST

Test case:
#include <avr/interrupt.h>
#include <compat/deprecated.h>

ISR(INT0_vect,ISR_NOBLOCK)
{
volatile long long foo;

foo = 1;
}

INTERRUPT(PCINT0_vect)
{
volatile long long foo;

foo = 1;
}

int main(void)
{
return 0;
}

For 'PCINT0_vect' interrupt handler correct code is generated:

INTERRUPT(PCINT0_vect)
{
8c:	78 94       	sei     // enable interrupt
....
//  Correct. interrupt is disabled/enabled when SP register is changed
9e:	cd b7       	in	r28, 0x3d	; 61
a0:	de b7       	in	r29, 0x3e	; 62
a2:	28 97       	sbiw	r28, 0x08	; 8
a4:	f8 94       	cli
a6:	de bf       	out	0x3e, r29	; 62
a8:	78 94       	sei
aa:	cd bf       	out	0x3d, r28	; 61

For 'INT0_vect' interrupt handler wrong code is generated:

ISR(INT0_vect,ISR_NOBLOCK)
{
48:	78 94       	sei     // enable interrupt
....
//  WRONG! NO interrupt is disabled/enabled when SP register is changed
5a:	cd b7       	in	r28, 0x3d	; 61
5c:	de b7       	in	r29, 0x3e	; 62
5e:	28 97       	sbiw	r28, 0x08	; 8
60:	de bf       	out	0x3e, r29	; 62
62:	cd bf       	out	0x3d, r28	; 61

Anatoly.

This issue was migrated from https://savannah.nongnu.org/bugs/?26734

avrs-admin, 2022-01-30:
Anatoly Sokolov <aesok>
Wed 03 Jun 2009 05:07:00 AM CEST

The ISR(INT0_vect,ISR_NOBLOCK) macro expand in:
void **_vector_1 (void) __attribute_** ((signal,used, externally_visible)) **_attribute_**((interrupt)); void __vector_1 (void)

This code confused the GCC. The 'interrupt' attribute makes emit the 'sei' instruction but the 'signal' attribute makes don't disable interrupt when SP reg is changed.

This is bad use 'interrupt' and 'signal' attribute simultaneously.

1. I will add in the GCC code for checking incorrect attributes use. i.e use 'interrupt' and 'signal' attribute simultaneously for one function.

2. avr-libc should be modified and don't use 'interrupt' and 'signal' attribute simultaneously.

Anatoly.

avrs-admin, 2022-01-30:
Georg-Johann Lay <gjlayde>
Sat 09 Jun 2012 12:45:18 AM CEST

With avr-gcc 4.7.1 and higher, attribute interrupt will override attribute signal, see

[http://gcc.gnu.org/PR53256](https://savannah.nongnu.orghttp://gcc.gnu.org/PR53256)

and the GCC manual.

sprintersb, 2024-02-17:
According to the last comment, this was added as an extension to GCC v4.7.1: When both `interrupt` and `signal` are specified, then the compiler assumes `interrupt`.

The issue itself was introduced with the `ISR` macros and the `ISR_NOBLOCK` attribute, because `ISR_NOBLOCK` adds `interrupt` but does not remove `sigal`.  This behaviour persists to date.

closed 2024-02-17 by sprintersb