avrdudes/avr-libc #436: [bug #30091] Header file generation script could generate garbage

avrs-admin, 2022-01-30:
Wed 09 Jun 2010 08:42:04 AM CEST

See **[bug #30018](https://savannah.nongnu.org/bugs/?30018)**: the (XSL) script that generates the header files
leaves stray characters (like "/" or "(") in macro names and/or
expansions.

I doubt the current "Device pin definitions" in iotn40.h et al.
are usable in any way.

This issue was migrated from https://savannah.nongnu.org/bugs/?30091

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Thu 10 Jun 2010 12:41:26 AM CEST

Another thing that needs to be changed in the convertor script is to be able to generate double-word registers.

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Wed 23 Feb 2011 09:02:48 PM CET

Postponed until AVR Studio 5 is released and a new XML format.

sprintersb, 2025-04-17:
Closed: We no more generate device headers these days.

closed 2025-04-17 by sprintersb