avrdudes/avr-libc #476: [bug #32945] Wrong PF implementation in pmstring

avrs-admin, 2022-01-30:
Tue 29 Mar 2011 08:05:42 PM CEST

It looks like that the pmstring code uses the LPM_R0_ZPLUS_* macros von common/macros.inc, which will only use the 'elpm' if **_AVR_ENHANCED_** is set. Unfortunately this is no longer used??

Therefore memcmp_PF etc. don't work with atmega1284p or probably other 128k devices.

As I'm not familiar with the setting of flags for the family avr51 I have no idea for a path

[file #29336: restore-rampz.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29336)

This issue was migrated from https://savannah.nongnu.org/bugs/?32945

avrs-admin, 2022-01-30:
Jan Waclawek <wek>
Thu 15 Sep 2011 07:43:33 PM CEST

This appears to be a consequence of the problem reported in [https://savannah.nongnu.org/patch/?6352](https://savannah.nongnu.orghttps://savannah.nongnu.org/patch/?6352) (thus duplicate).

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Mon 26 Sep 2011 07:10:25 PM CEST

I've changed the LPM_R0_ZPLUS_* macros in common/macros.inc to look for **_AVR_HAVE_ELPM_** instead of BIG_CODE.

My guess is that there are still problems because of finding the definition of AVR_RAMPZ_ADDR.

To really fix this, avr-libc needs to build a library-per-device.

avrs-admin, 2022-01-30:
Georg-Johann Lay <gjlayde>
Tue 08 Oct 2013 09:20:05 PM CEST

If someone fixes this issue, please don't forget to reset RAMPZ to 0 after changing it.  This is necessary for devices that have RAMPD.  Otherwise, reading vom RAM using indirect Z addressing might read from the wrong address.  For devices without RAMPD, RAMPZ need not to be reset to 0.

Cf. also the notes on RAMPx in the GCC docs at [http://gcc.gnu.org/onlinedocs/gcc/AVR-Options.html](https://savannah.nongnu.orghttp://gcc.gnu.org/onlinedocs/gcc/AVR-Options.html)

avrs-admin, 2022-01-30:
Pitchumani <pitchumani>
Wed 09 Oct 2013 11:58:51 AM CEST

> **[comment #2](https://savannah.nongnu.org#comment2)**
> I've changed the LPM_R0_ZPLUS_* macros in common/macros.inc to look for **AVR_HAVE_ELPM** instead of BIG_CODE.
>My guess is that there are still problems because of finding the definition of AVR_RAMPZ_ADDR.

RAMPZ register address is defined based on RAMPZ macro in device header file. Building library for specific arch will result in error as AVR_RAMPZ_ADDR is not defined.

(include/avr/common.h)
284 #if defined(RAMPZ)
285 #  define AVR_RAMPZ_REG   RAMPZ
286 #  if **_AVR_ARCH_** >= 100
287 #    define AVR_RAMPZ_ADDR  _SFR_MEM_ADDR(RAMPZ)
288 #  else
289 #    define AVR_RAMPZ_ADDR  _SFR_IO_ADDR(RAMPZ)
290 #  endif
291 #endif

It is possible to hard-code RAMPZ register address irrespective of device header file (gcc also hard-codes RAMPZ address) as follows:

+#  define AVR_RAMPZ_REG   (_SFR_IO8(0x3b))
+#  define AVR_RAMPZ_ADDR  (_SFR_IO8(0x3b))

This will enable building library for a architecture than a device.
--
This fix shall be included until we have per device library build system - Joerg Wunsch ([http://lists.nongnu.org/archive/html/avr-libc-dev/2013-10/msg00002.html](https://savannah.nongnu.orghttp://lists.nongnu.org/archive/html/avr-libc-dev/2013-10/msg00002.html))

avrs-admin, 2022-01-30:
Pitchumani <pitchumani>
Wed 09 Oct 2013 03:08:19 PM CEST

RAMPZ address is hard-coded already (revision 2279) in case of building avr-libc. This resolves AVR_RAMPZ_ADDR definition error.

I have attached patch (restore-rampz.patch) to restore RAMPZ value for all xxx_PF() functions.

ChangeLog
2013-10-09  Pitchumani Sivanupandi <pitchumani.s@atmel.com>

Fix for **[bug #32945](https://savannah.nongnu.org/bugs/?32945)**.

- common/macros.inc: Save RAMPZ register if RAMPD is present.Add LPM_R0_ZPLUS_FINI macro to restore RAMPZ if RAMPD exists.

- libc/pmstring/memcmp_PF.S: Restore RAMPZ before return.- libc/pmstring/memcpy_PF.S: Same.- libc/pmstring/strcasecmp_PF.S: Same.- libc/pmstring/strcat_PF.S: Same.- libc/pmstring/strcmp_PF.S: Same.- libc/pmstring/strcpy_PF.S: Same.- libc/pmstring/strlcat_PF.S: Same.- libc/pmstring/strlcpy_PF.S: Same.- libc/pmstring/strlen_PF.S: Same.- libc/pmstring/strncasecmp_PF.S: Same.- libc/pmstring/strncat_PF.S: Same.- libc/pmstring/strncmp_PF.S: Same.- libc/pmstring/strncpy_PF.S: Same.- libc/pmstring/strnlen_PF.S: Same.- libc/pmstring/strstr_PF.S: Same.Please review.

(**[file #29336](https://savannah.nongnu.org/support/download.php?file_id=29336)**)

sprintersb, 2024-03-16:
https://github.com/avrdudes/avr-libc/commit/a0d37255bbfb50714d8bae2f78adaf0731f5bcbd fixed the last bit of this issue: Restore `RAMPZ` as needed, i.e. on EBI devices. On devices without EBI, `RAMPZ` need not to be restored (or such devices don't have `RAMPZ` to begin with).

closed 2024-03-16 by sprintersb