avrdudes/avr-libc #482: [bug #33716] pgmspace typedefs not legal

avrs-admin, 2022-01-30:
Tue 05 Jul 2011 11:48:34 AM CEST

Apparently, "__attribute__((progmem))" is not supported by avr-gcc in association with types, but only with actual variables.  This makes about half of avr/pgmspace.h (and associated documentation?) useless (or at least subject to becoming useless at any time.)

see [http://gcc.gnu.org/bugzilla/show_bug.cgi?id=38342](https://savannah.nongnu.orghttp://gcc.gnu.org/bugzilla/show_bug.cgi?id=38342)

This issue was migrated from https://savannah.nongnu.org/bugs/?33716

avrs-admin, 2022-01-30:
Georg-Johann Lay <gjlayde>
Sat 09 Jul 2011 12:25:31 PM CEST

As Bill already wrote, attribute progmem just works by accident in avr-gcc ( and not at all in avr-g++).

It appears that
const prog_char c = 'a';
works for C and not for C++.  However, there is **no** support for progmem in types it in avr-gcc and there is no guarantee that it will work for any code imagineable.

I highly recommend to deprecate types like prog_char:

typedef char **_attribute_**((progmem,deprecated)) prog_char;

or

typedef char **_attribute_**((progmem,deprecated("message"))) prog_char;

Notice that despite "progmem", "deprecated" **is** documented for types.

As the next release of avr-libc will come with bulk of poison, anyway, this bit of deprecation won't hurt and will help users to write more robust code.

Johann

avrs-admin, 2022-01-30:
Eric Weddington <arcanum>
Mon 26 Sep 2011 05:46:45 PM CEST

Committed changes in pgmspace.h to deprecate the typedefs with **_progmem_** attribute, and documentation to match.

Need to:
- Check if any other places need deprecation.
- See if other places in avr-libc use the deprecated typedefs and fix accordingly.

avrs-admin, 2022-01-30:
Jan Waclawek <wek>
Mon 26 Sep 2011 08:48:12 PM CEST

Rather than deprecating the typedefs in pgmspace.h, I would suggest to replace them by respective macros.

The reason for this is, that those typedefs are provided as a convenience for **users**, who would like to have the simplest possible means for placing variables into the code memory (and maybe also as a basis for writing universal code which could be migrated between different AVR C compilers). I accept the fact that the progmem typedefs are broken and the developers refuse to fix them, and I accept it in the hope that the real named spaces will be available soon. Those typedefs/macros would then provide also a seamless migration path to the new way of placing/accessing such variables.

Otherwise, a large amount of existing code - including examples in the documentation - will start to throw those warnings quite unnecessarily.

I here repeat what I've already said in [http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=871000#871000](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=871000#871000)

Jan Waclawek

sprintersb, 2024-02-17:
Fixed by e617964a8dc650069cdb24a97ed69f4a2e1 and 3045e509336a8cf4c6b1493f33ad512.

closed 2024-02-17 by sprintersb