avrdudes/avr-libc #503: [bug #35198] sleep.h missing include of avr/sfr_defs.h

avrs-admin, 2022-01-31:
Fri 30 Dec 2011 11:08:56 AM CET

sleep.h missing include of <avr/sfr_defs.h>

As per Jörg's request , this is filed as a bug report.
See [http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=580767#580767](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=580767#580767)

/Bingo

[file #25195: sleep.h.patch](https://savannah.nongnu.org/bugs/download.php?file_id=25195)

This issue was migrated from https://savannah.nongnu.org/bugs/?35198

avrs-admin, 2022-01-31:
Dmitry Xmelkov <dmix>
Mon 16 Jan 2012 12:48:48 PM CET

Hmm...
'avr/sleep.h' includes 'avr/io.h'.
The last includes 'avr/sfr_defs.h'.
IMHO, all is correct.

avrs-admin, 2022-01-31:
Bingo <bingo>
Mon 27 Feb 2012 08:07:18 PM CET

Well as <avr/io.h> is included sfr_defs.h would be also.
I didn't check as Jörg sait it was a but that it wasn't included

But i was bitten by the __BV bug in avr/sleep.h  (90S8515 section)

Attached is a patch

/Bingo

(**[file #25195](https://savannah.nongnu.org/support/download.php?file_id=25195)**)

avrs-admin, 2022-01-31:
Eric Weddington <arcanum>
Fri 16 Nov 2012 07:50:00 PM CET

Joerg, you seem to know the history surrounding this bug....

Assigned to you.

sprintersb, 2024-07-31:
Fixed by e8fd553a0ffa8b4acd050be080ad57756f74e203.

closed 2024-07-31 by sprintersb