avrdudes/avr-libc #510: [bug #35399] User manual on website has incomplete alphabetical index

avrs-admin, 2022-01-31:
Sun 29 Jan 2012 11:08:15 PM CET

avr-libc user manual on website has incomplete alphabetical index:

[http://www.nongnu.org/avr-libc/user-manual/globals.html](https://savannah.nongnu.orghttp://www.nongnu.org/avr-libc/user-manual/globals.html)

Something must be up with doxygen generation...

This issue was migrated from https://savannah.nongnu.org/bugs/?35399

sprintersb, 2025-04-29:
As is seems, the index is generated, but is never linked from anywhere.

https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_a.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_b.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_c.html
...
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_func.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_func_a.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_func_b.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_func_c.html
...
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_defs.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_defs_a.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_defs_b.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_defs_c.html
...
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_vars.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_type.html
https://avrdudes.github.io/avr-libc/avr-libc-user-manual/globals_enum.html

sprintersb, 2025-04-30:
Added a global index (by hand).