avrdudes/avr-libc #513: [bug #35483] ./configure does not accept --target

avrs-admin, 2022-01-31:
Wed 08 Feb 2012 05:14:26 PM CET

I didn't build avr-libc so I don't know when this (mis)feature appeared...

The top level configure script requires to spell out --host=avr. IMHO this is unfortunate for several reasons:

1) the product is called avr-libc. I wouldn't expect it to provide anything else (e.g. a libc for ARM) than a libc for avr. Nevertheless you have to type that option altough the only permitted value is 'avr*'

2) It is a library for my 'target', not for my 'host'.  The hosting machine typically is a Linux or Windows PC.  That's why I stumbled over this issue in the first place. I naivly tried "./configure --target=avr" as for binutils and gcc and it failed.  (I can vaguely imagine a logic why it is called host, but that's very counterintuitive!)

3) I had a third reason when I opened the issue but I don't remember anymore...

I propose to accept --target= as an equivalent to --host in the first step, correct the build instructions and then to deprecate --host as a target indicator (it might still be useful for an alternative name of --build in case config.guess fails to correctly detect the hosting machine)

Rolf

This issue was migrated from https://savannah.nongnu.org/bugs/?35483

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Wed 08 Feb 2012 11:57:15 PM CET

All that stuff still originates from Ted Roth's initial work on the
autoconf setup (back in 2004, the first revision appeared on branch
"autoconf-branch", it's r498 now after the migration to SVN).
Frankly, I've got no idea why he required this scheme, but I'm pretty
sure he's got a reason by that time. ;-)

If someone could come up with a better implementation that produces
similar results, I'm all ears.

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Fri 01 Feb 2013 09:52:24 PM CET

--host specifies the platform that runs the software, thus --target=avr would be confusing.

--host is common practice.  For example, if you cross-build libraries like GSL (so that they run on a platform fifferent to the build platform), you specify that by means of --host.

avrs-admin, 2022-01-31:
Mike Frysinger <vapier>
Sun 30 Jun 2013 08:00:26 AM CEST

this bug is invalid and should be closed.  avr-libc is behaving correctly.

for documentation on the arguments, please see:
 [https://www.gnu.org/savannah-checkouts/gnu/autoconf/manual/autoconf-2.69/html_node/Specifying-Target-Triplets.html#Specifying-Target-Triplets](https://savannah.nongnu.orghttps://www.gnu.org/savannah-checkouts/gnu/autoconf/manual/autoconf-2.69/html_node/Specifying-Target-Triplets.html#Specifying-Target-Triplets)

avrs-admin, 2022-01-31:
Josef Wolf <joedoe>
Tue 26 May 2020 01:49:28 PM CEST

**[comment #3:](https://savannah.nongnu.org#comment3)**

> this bug is invalid and should be closed.  avr-libc is behaving correctly.

Sorry, I tend to disagree: it is NOT behaving correctly.

> for documentation on the arguments, please see: [...]

According to this documentation, there are up to three systems involved with cross-compilers:

--build specifies the system which builds the toolchain.

--host specifies the system which executes the toolchain.

--target specifies the system which executes the code that the toolchain generated.

Thus, when building a toolchain on a linux which is meant to run on windows to generate code for avr, the correct specification would be:

--target=avr --build=x86_64-pc-linux-gnu --host=x86_64-pc-windows

or something. After all, the AVR is not executing the toolchain. It is executing the code which is generated by the toolchain.

The semantic of --host=avr is to build a toolchain which is running on the AVR. I doubt that anybody is really trying to do such a thing...

Every other tool in the world specifies it this way.

BTW: while at it: the configure script also don't accept the target as a TRIPLET. That is: configure will fail when the target is specified as avr-XXX-YYY

avrs-admin, 2022-01-31:
Mike Frysinger <vapier>
Fri 29 May 2020 05:22:59 PM CEST

> According to this documentation, there are up to three systems involved with cross-compilers:
>
> --build specifies the system which builds the toolchain.
>
> --host specifies the system which executes [...]

you're misreading the targets, and your subsequent characterization of them is wrong.  windows is **not** executing avr-libc, nor is avr-libc producing code.  hence your your --host=windows & --target=avr-libc are incorrect.

> BTW: while at it: the configure script also don't accept the target as a TRIPLET. That is: configure will fail when the target is specified as avr-XXX-YYY

exactly: libc's do not generate code.  which is why --target isn't supported.

sprintersb, 2024-02-04:
From the [autoconf manual](https://www.gnu.org/software/autoconf/manual/autoconf-2.68/html_node/Specifying-Target-Triplets.html):

<dl>
<dt><code>--build=build-type</code>
<dd>the type of system on which the package is being configured and compiled. It defaults to the result of running config.guess.
<dt><code>--host=host-type</code>
<dd>the type of system on which the package runs. By default it is the same as the build machine. Specifying it enables the cross-compilation mode.
<dt><code>--target=target-type</code>
<dd>the type of system for which any compiler tools in the package produce code (rarely needed). By default, it is the same as host. 
</dl>

It is `--host` because the code like for `printf` is being run on AVR.

It's not `--target` because the AVR-LibC code like `printf` or `sin` isn't producing code to run on AVR.

closed 2024-02-04 by sprintersb