avrdudes/avr-libc #518: [bug #35948] iom32u4.h for ATmega32U4 incorrectly defines Timer 2

avrs-admin, 2022-01-31:
Thu 22 Mar 2012 04:44:05 PM CET

it's my understanding from the datasheet that the atmega32u4 doesn't have a Timer 2, so it seems like the definitions for TCCR2A, TIFR2, TCNT2, etc, should be missing from the header file.

all of those are, indeed, missing from iom32u2.h, so i think their presence in iom32u4.h must be just an oversight.

apparently timer 2 isn't mentioned in the ATmega32U4.xml part description file provided with AVR Studio, either, though i was told this at the avrfreaks forum, and don't know it firsthand.

(my version is 1.6.8-2, from ubuntu natty.)

paul

[file #27593: 35498.patch](https://savannah.nongnu.org/bugs/download.php?file_id=27593)

This issue was migrated from https://savannah.nongnu.org/bugs/?35948

avrs-admin, 2022-01-31:
Eric Weddington <arcanum>
Thu 28 Jun 2012 06:54:03 PM CEST

This bug affects Arduino project and needs to get fixed before next release.

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Thu 20 Mar 2014 04:00:53 PM CET

removed timer2 related definitions from iom32u4.h
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2413](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2413)

sprintersb, 2024-02-07:
Fixed by 8caae7e4c515f546c56ab985239181e47e0497d4.

closed 2024-02-07 by sprintersb