avrdudes/avr-libc #519: [bug #35971] attiny4313 (2313a) pin-change interrupts PCINT[0...2]_vect etc.

avrs-admin, 2022-01-31:
Sat 24 Mar 2012 12:50:40 AM CET

According to the datasheet these mcus support pin-change interrupts in 3 groups and hence also 3 pin-change interrupts:

PCINT0_vect ( vector #12 - avr-libc counting: 11 )
PCINT1_vect ( vector #20 -                    19 )
PCINT2_vect ( vector #21 -                    20 )

Strangely I can only find the enable bit 'PCIE' for group 0 in the latest 'iotn4313.h' I got from this site. It should have PCIE0, PCIE1 and PCIE2. However PCMSK2 for example is defined, but useless without the ISR enabled.

Also what is unfortunate is the naming convention of the interrupts.

PCINT0_vect is called PCINT_B_vect
PCINT1_vect is called PCINT_A_vect
PCINT2_vect is caleld PCINT_D_vect

Where's PCINT_C_vect btw?

Sticking to the names of the datasheet would be best.

This issue was migrated from https://savannah.nongnu.org/bugs/?35971

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Thu 20 Mar 2014 12:20:13 PM CET

interrupt number starts from 0 in avr-libc.
-
UCPHA, UDORD bit definitions missing for UCSRC
0x03A: should be GIFR, not EIFR

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Thu 20 Mar 2014 12:23:56 PM CET

fixed.
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2412](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2412)

sprintersb, 2024-02-07:
Fixed by d06f7aae0b438969162123c519160856f5b46b53.

closed 2024-02-07 by sprintersb