avrdudes/avr-libc #521: [bug #36124] Improve documentation on .bss clear when used in assembler

avrs-admin, 2022-01-31:
Sat 07 Apr 2012 04:02:23 PM CEST

Newer builds of the avr-gcc toolchain, such as WinAVR20100110, or builds including the PR18145 fixup (4.7+), contain a patch which results in conditional linking of the .bss cleanup loop only if it is needed for the C source (similarly for the initialized .data). This may lead to surprising behaviour when such variables are defined only in assembler parts of a project.

It is therefore suggested to add something like the following text to [http://www.nongnu.org/avr-libc/user-manual/assembler.html](https://savannah.nongnu.orghttp://www.nongnu.org/avr-libc/user-manual/assembler.html) after "For devices that have RAM, initializing the RAM variables can also be utilized.", or [http://www.nongnu.org/avr-libc/user-manual/mem_sections.html#sec_dot_init](https://savannah.nongnu.orghttp://www.nongnu.org/avr-libc/user-manual/mem_sections.html#sec_dot_init) into the chapters dealing with .data and .bss sections, or both as appropriate:

Note: To ensure that variables in .bss section are zeroed upon startup, add the following line to the assembler source:
.global __do_clear_bss

To ensure that variables in .data section are initialized from FLASH during startup, add the following line to the assembler source:
.global __do_copy_data

(This submission is result of discussion on AVRFreaks [http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&t=119131&highlight](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&t=119131&highlight)= )

This issue was migrated from https://savannah.nongnu.org/bugs/?36124

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Tue 20 Nov 2012 07:02:09 PM CET

Since GCC 3.4, the respective bits of startup code are hosted with GCC (namely as part of libgcc) and not in AVR Libc.

It's highly recommended that any user who updates or downgrades or changes vendor / distributor of the toolchain reads all the relevent release notes.

This includes the GCC release notes, of course.

If you change the distributor, it's also strongly recommended to get informed about differences in the binary interface (ABI), because some private ports like Atmel use different ABI.

Bottom line: I think a pointer to the GCC release notes in the AVR Libc documentation is reasonable, but I don't think it makes much sense to copy paste the GCC release notes to the AVR Libc docs...

For the GCC 4.7 release notes, see
 [http://gcc.gnu.org/gcc-4.7/changes.html](https://savannah.nongnu.orghttp://gcc.gnu.org/gcc-4.7/changes.html)

For the GCC 4.6 release notes, see
 [http://gcc.gnu.org/gcc-4.6/changes.html](https://savannah.nongnu.orghttp://gcc.gnu.org/gcc-4.6/changes.html)

For the avr-gcc ABI, see [http://gcc.gnu.org/wiki/avr-gcc](https://savannah.nongnu.orghttp://gcc.gnu.org/wiki/avr-gcc)

For vendor-specific release notes and changes to the ABI please consult respective document of your vendor.

sprintersb, 2024-08-02:
Done.

closed 2024-08-02 by sprintersb