avrdudes/avr-libc #526: [bug #36571] stdint.h: INTn_C not standard compliant

avrs-admin, 2022-01-31:
Thu 31 May 2012 10:02:57 PM CEST

C99: 7.18.4 Macros for integer constants

writes in clause #3 on the INTn_C function-like macros:

Each invocation of one of these macros shall expand
to an integer constant expression suitable for use
in #if preprocessing directives.

However, stdint.h defines respective macros like so

#define INT8_C(value) ((int8_t) value)

which cannot be used in #if preprocessing directives.

FYI, avr-gcc built-in defines similar macros like so, provided -mint8 is off:

$ echo | avr-gcc -x c - -E -dM | sort | grep 'INT.*_C'

#define __INT8_C(c) c
#define __INT16_C(c) c
#define __INT32_C(c) c ## L
#define __INT64_C(c) c ## LL
#define __INTMAX_C(c) c ## LL
#define __UINT8_C(c) c
#define __UINT16_C(c) c ## U
#define __UINT32_C(c) c ## UL
#define __UINT64_C(c) c ## ULL
#define __UINTMAX_C(c) c ## ULL

With -mint8 on, it should be something around (no appropriate compiler handy at the moment)

#define __INT8_C(c) c
#define __INT16_C(c) c ## L
#define __INT32_C(c) c ## LL
#define __INTMAX_C(c) c ## LL
#define __UINT8_C(c) c ## U
#define __UINT16_C(c) c ## UL
#define __UINT32_C(c) c ## ULL
#define __UINTMAX_C(c) c ## ULL

So avr-libc could adopt these or just

#define INT8_C(c) __INT8_C(c)

which works since gcc 4.5, maybe also 4.4.

This issue was migrated from https://savannah.nongnu.org/bugs/?36571

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Sat 17 Nov 2012 12:00:00 AM CET

Here is the output for -mint8:

$ echo | avr-gcc-4.7.2 -x c - -E -dM -mint8 | sort | grep 'INT.*_C'

#define __INT16_C(c) c ## L
#define __INT32_C(c) c ## LL
#define __INT8_C(c) c
#define __INTMAX_C(c) c ## LL
#define __UINT16_C(c) c ## UL
#define __UINT32_C(c) c ## ULL
#define __UINT8_C(c) c ## U
#define __UINTMAX_C(c) c ## ULL

Notice that simply adopting this does not work with macros.  Just consider:

#define F_CPU 0x80000

long f (void)
{
return INT32_C (F_CPU);
}

which works with

#define INT32_C(c) __INT32_C(c)

but fails with

#define INT32_C(c) c ## L

sprintersb, 2024-02-04:
Fixed by 889d3ff66701e11d688af4efea71078dad272064.

closed 2024-02-04 by sprintersb