avrdudes/avr-libc #529: [bug #36611] _delay_loop_2 must not use constraint "w" with all targets

avrs-admin, 2022-01-31:
Sat 09 Jun 2012 12:36:27 AM CEST

./include/util/delay_basic.h

implements:

void
_delay_loop_2(uint16_t __count)
{
 **_asm_** volatile (
"1: sbiw %0,1" "\n\t"
"brne 1b"
: "=w" (__count)
: "0" (__count)
);
}

This won't work for tiny tiny targets because

- they don't have SBIW
- compiler with suprt for these targets empty the register
class associated with "w".  Thus, the compiler will throw
an error if it sees "w".

This issue was migrated from https://savannah.nongnu.org/bugs/?36611

avrs-admin, 2022-01-31:
Eric Weddington <arcanum>
Fri 16 Nov 2012 07:13:05 PM CET

Johann, what do you recommend to fix this?

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Fri 16 Nov 2012 10:04:06 PM CET

Factor out the ISA / architecture by means of #if / #ifdef **_HAVE_ADIW_** or **_HAVE_CONSTRAINT_W_** or **_AVR_ARCH_** or whatever your tools provide and use 2-insn decrement alongside with "d".

If your tools have __builtin_delay_cycles you can use that, but make sure that respective insn constraint does not suffer from the same flaw (AFAIR this is the case with Atmel patches)

avrs-admin, 2022-01-31:
Eric Weddington <arcanum>
Fri 16 Nov 2012 10:17:17 PM CET

We'll need to verify that __builtin_delay_cycles in gcc is still valid.

avrs-admin, 2022-01-31:
Cameron Tacklind <cinderblock>
Wed 10 May 2017 03:45:07 AM CEST

I ran into this problem and just re-wrote the ****_delay_loop_2**** assembly.

This should be equivalent for all AVRs. It should even take the same number of clock cycles (since ****sbiw**** takes 2 clock cycles).

My initial tests suggest this is working. Since ****__count**** is a 2 byte number, ****%A0**** and ****%B0**** should always work as expected.

```
void
_delay_loop_2(uint16_t __count)
{
__asm__ volatile (
"1: subi %A0,1" "\n\t"
"sbci %B0,0" "\n\t"
"brne 1b"
: "=d" (__count)
: "0" (__count)
);
}

```

sprintersb, 2024-02-04:
Dup of #902, #822.

closed 2024-02-04 by sprintersb