avrdudes/avr-libc #542: [bug #37778] _MemoryBarrier() in cpufunc.h error on compile

avrs-admin, 2022-01-31:
Mon 19 Nov 2012 10:52:10 PM CET

The _MemoryBarrier() macro defined in avr/cpufunc.h results in the following error when compiling:
error: expected string literal before ':' token

Currently defined as:
#define **MemoryBarrier() __asm_** __volatile__(:::"memory")

Should be:
#define **MemoryBarrier() __asm_** __volatile__("":::"memory")

[file #30890: pr37778.patch](https://savannah.nongnu.org/bugs/download.php?file_id=30890)

This issue was migrated from https://savannah.nongnu.org/bugs/?37778

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Mon 19 Nov 2012 11:59:22 PM CET

...and while your are fixing avr/cpufunc.h, don't forget to fix _NOP which reads:

#define **NOP() __asm_** __volatile__("nop")

but should get a memory barrier:

#define **NOP() __asm_** __volatile__("nop" ::: memory)

avrs-admin, 2022-01-31:
Dmitry Xmelkov <dmix>
Sun 27 Jan 2013 01:58:16 PM CET

What is the necessity to add memory barrier for _NOP()?
Common usage, like:
PORTB |= 1;
_NOP();
PORTB &= ~1;
is safe, as hardware registers are volatile.

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Sun 27 Jan 2013 06:28:17 PM CET

Question is whether or not NOP should be allowed to be moved across
ordinary memory moves.

For SEI and CLI, for example, I think this is not wanted in general.

NOP is less intrusive than SEI / CLI, of course, but avr-gcc emits
__builtin_avr_nop() with a memory clobber:

[http://gcc.gnu.org/viewcvs/trunk/gcc/config/avr/avr.md?revision=195151&view=markup#l5938](https://savannah.nongnu.orghttp://gcc.gnu.org/viewcvs/trunk/gcc/config/avr/avr.md?revision=195151&view=markup#l5938)

sprintersb, 2024-02-07:
Fixed by 5f32fb6a729e5fb45c14fdb8cc83d4367ea60a3b.

closed 2024-02-07 by sprintersb