avrdudes/avr-libc #545: [bug #37847] Xmega power.h groupings are wrong

avrs-admin, 2022-01-31:
Sat 01 Dec 2012 01:34:10 AM CET

In working on a patch to add **all** xmega devices, I ran across the power.h code.  The problem is that the groupings don't actually match the functionality of the various devices, and even within those groups some of the macros aren't consistent.

I'd propose that rather than defining them based on the individual chip macros, that each of the power_*_enable() and power_*_disable() macros be #ifdef'd based on whether the peripheral itself is defined:

#ifdef USARTF1
#define power_usartf1_enable()  (PR_PRPF &= (uint8_t)~(PR_USART1_bm))
#define power_usartf1_disable() (PR_PRPF |= (uint8_t)PR_USART1_bm)
#endif

The _enable_all and _disable_all are a slightly tougher nut to crack, but that could be solved by adding in before the #endif a:

#define __power_usartf1_mask ((uint8_t)PR_USART1_bm)
#else
#define __power_usartf1_mask (0)

Then grouping all these macros together into a single pair of definitions for _all().

[file #27005: power.h](https://savannah.nongnu.org/bugs/download.php?file_id=27005)

This issue was migrated from https://savannah.nongnu.org/bugs/?37847

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Mon 09 Mar 2015 07:41:28 AM CET

This is fixed by **[patch #8563](https://savannah.nongnu.org/patch/?8563)**.
 [https://savannah.nongnu.org/patch/?func=detailitem&item_id=8563](https://savannah.nongnu.orghttps://savannah.nongnu.org/patch/?func=detailitem&item_id=8563)

sprintersb, 2025-09-14:
Fixed by 141ebec51ef8e7acbf0695fa7e4225e699830095 and 82dce0947d36e7ca1aeb0c620ea104283ba770ab.

closed 2025-09-14 by sprintersb