avrdudes/avr-libc #553: [bug #38125] Distribute gcrt1.S

avrs-admin, 2022-01-31:
Sun 20 Jan 2013 04:15:31 PM CET

The assembler source code of the startup code  ./crt1/gcrt1.S -n could be added to the installation.

This makes it easier for the user to adjust the startup code and / or compiler it for devices that are not yet supported by avr-gcc as lined out in [http://gcc.gnu.org/wiki/avr-gcc#Supporting_.22unsupported.22_Devices avr-gcc Wiki: Supporting "unsupported" Devices].

The onyl dependency should be #include <avr/io.h>.

Other dependencies like macros.inc and sectionname.h can easily be avoided altogether, which makes using gcrt1.S more straigh forward.

This issue was migrated from https://savannah.nongnu.org/bugs/?38125

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Tue 12 Aug 2014 12:27:46 AM CEST

Do you have a suggestion about **where** to install it?

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Mon 25 Aug 2014 06:04:48 PM CEST

I'd propose in a place that's part of the default include path so that it can be dragged by #include "<optional-path>/gcrt1.S". Maybe in ./avr/include/util ?

Unfortunately gcrt1.S has some dependencies (macros.inc) and some awkward includes (#inclule IOSYMFILE) whith make it harder to (re)use the startup code.

sprintersb, 2024-03-22:
Imo this is superseeded now by device-packs. The include requirements in `gcrt1.S`  are now evel less user friendly than they were back then.

closed 2024-03-22 by sprintersb