avrdudes/avr-libc #556: [bug #38516] Missing TWI and UCSR1D definitions for ATmega16/32 U4

avrs-admin, 2022-01-31:
Thu 14 Mar 2013 03:00:40 PM CET

avr/iom16u4.h does not contain TWI Control register addresses and bits definitions.

#define TWCR    _SFR_IO8(0x36)
#define TWIE    0
#define TWEN    2
#define TWWC    3
#define TWSTO   4
#define TWSTA   5
#define TWEA    6
#define TWINT   7

[file #30860: DEVXML-176_iom16u4-twi.patch](https://savannah.nongnu.org/bugs/download.php?file_id=30860)

This issue was migrated from https://savannah.nongnu.org/bugs/?38516

avrs-admin, 2022-01-31:
Pasi Heinonen <pasihe>
Thu 14 Mar 2013 08:49:26 PM CET

Comparing to datasheet the definitions are incomplete.
I will send a patch.

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Thu 20 Mar 2014 10:01:01 AM CET

Below TWI register definitions also missing in iom16u4.h
TWBR, TWSR, TWAR, TWDR, TWAMR

UCSR1D register definition also missing in iom16u4.h

Patch attached by Morten fixes this.

sprintersb, 2024-02-07:
Fixed by e6be0368f2810a68490060279047453f549b6d76.

closed 2024-02-07 by sprintersb