avrdudes/avr-libc #563: [bug #39008] Progmem macros do not work for TINY architecture devices

avrs-admin, 2022-01-31:
Thu 16 May 2013 02:39:58 PM CEST

The macros in pgmspace.h do not work for TINY devices (attiny10/20/40 etc.), as they do not have any variant of the LPM instruction.

Instead, flash memory is mapped to data memory space at a certain offset, and normal data memory load instructions can be used to read from flash.

The attached patch customizes the **_LPM series of macros for __AVR_TINY_**, reimplementing them using plain LD, with the memory offset coming from a macro (**_AVR_TINY_PM_BASE_ADDRESS_**) defined by the compiler.

Changelog

2013-05-16  Pitchumani Sivanupandi <pitchumani.s@atmel.com>

* include/avr/pgmspace.h: Customize and reimplement __LPM
macros for tiny architecture.

[file #28096: lpm_tiny.patch](https://savannah.nongnu.org/bugs/download.php?file_id=28096)

This issue was migrated from https://savannah.nongnu.org/bugs/?39008

sprintersb, 2023-08-01:
IMO we should remove any LPM / pgm_read stuff for reduced Tiny altogether.

You can just write plain C code.  const in static storage will be allocated to `.rodata` which is located in `.text` (flash) and **not** in RAM like with all other AVRs (except avrxmega3 which also have `.rodata` in flash).

You might also want to read [avr-gcc's documentation on `progmem`](https://gcc.gnu.org/onlinedocs/gcc-8.5.0/gcc/AVR-Variable-Attributes.html#AVR-Variable-Attributes), in particular the part concerning reduced Tiny:

As of what I wrote above, you can drop `progmem` / `pgm_read_*` altogether and resort to vanilla C without loss of performance.

If you insist on using `progmem`, just use it.  There is **no need for special accessors** though, as avr-gcc will add according offset of 0x4000 to all symbols attributed `progmem`. For example, when you
```c++
__attribute__((__progmem__))
const char str[] = "Hallo.";

const char* get_addr (void)
{
    return str + 2;
}
```

`avr-gcc -mmcu=avrtiny ...` will compile this to

```
get_addr:
	ldi r24,lo8(str+16386)
	ldi r25,hi8(str+16386)
	ret
```
Notice 16386 = 0x4000 + 2.

Bottom line is that AVR-LibC should point this out or maybe should refer to avr-gcc's documentation.

leonty, 2023-10-02:
Removing LPM/pgm_read at all would reduce the code portability. Somebody would probably write code/firmware compatible with wide set of attiny/atmega MCUs, in this case a uniform approach to address the flash memory would be preferable.
The patch above doesn't introduce any additional restrictions and offers more portability at the user code level.

sprintersb, 2023-10-02:
>  The patch above doesn't introduce any additional restrictions and offers more portability at the user code level.

As  I already exlpained, **that patch is wrong**. If, at all, it would go something like

```c++
#define __LPM_tiny__(addr)  (*(addr))
```

sprintersb, 2024-02-15:
Fixed by https://github.com/avrdudes/avr-libc/commit/1e1a57720a6e1156d3134e7631c70198d212f244