avrdudes/avr-libc #565: [bug #39049] Clock prescaler set and get are missing for TINY architecture

avrs-admin, 2022-01-31:
Thu 23 May 2013 06:54:01 AM CEST

Implementation for clock_prescaler_set and clock_prescaler_get is missing for TINY architecture devices. The attached patch adds them to power.h

ChangeLog

2013-05-23  Usha Gupta <usha.gupta@atmel.com>

* include/avr/power.h: Implement clock_prescale_get and set for
tiny architecture devices.

[file #28144: tiny_clock_prescale.patch](https://savannah.nongnu.org/bugs/download.php?file_id=28144)

This issue was migrated from https://savannah.nongnu.org/bugs/?39049

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Tue 29 Apr 2014 12:34:54 PM CEST

committed the changes.
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2422](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2422)

This patch has been applied and committed.

Thank you for the contribution.

sprintersb, 2024-02-07:
Fixed by 72eaf005f9b5be6e84aef3592a0bf360e10755cb.

closed 2024-02-07 by sprintersb