avrdudes/avr-libc #566: [bug #39061] Clock_prescale_get and set missing for some mega and tiny devices

avrs-admin, 2022-01-31:
Fri 24 May 2013 10:10:22 AM CEST

The implementation for clock_prescale_set and get is missing for the following devices.

ATmega168A, ATmega16M1, ATmega16U2, ATmega324PA, ATmega3290P, ATmega64M1, ATmega64C1, ATmega88A, ATmega88PA, ATtiny88 and ATtiny87

The attached patch adds support by adding conditional directives for these devices to an existing implementation.

ChangeLog

2013-05-24  Usha Gupta <usha.gupta@atmel.com>

* include/avr/power.h: Add clock_prescale_get and set for
ATmega168A, ATmega16M1, ATmega16U2, ATmega324PA, ATmega3290P,
ATmega64M1, ATmega64C1, ATmega88A, ATmega88PA, ATtiny88 and
ATtiny87

[file #28157: misc_power.patch](https://savannah.nongnu.org/bugs/download.php?file_id=28157)

This issue was migrated from https://savannah.nongnu.org/bugs/?39061

avrs-admin, 2022-01-31:
Andrew Mao <mizzao>
Tue 02 Dec 2014 07:13:42 AM CET

I just ran into this issue as well using an ATmega16U2, and would love to see it fixed!

sprintersb, 2024-03-16:
Applied as https://github.com/avrdudes/avr-libc/commit/d0f316ae8499cede40ebcd0f6798b52f05ae4a99.

closed 2024-03-16 by sprintersb