avrdudes/avr-libc #569: [bug #39783] CRC missing definitions and incorrect power macros for xmega D3 devices

avrs-admin, 2022-01-31:
Wed 14 Aug 2013 08:54:04 AM CEST

- CRC module macro definitions are missing for xmega 64/128/192/256 D3 devices.- Power reduction macro definitions are incorrect for xmega 64/128/192 D3 devices.- Power reduction definitions are missing for xmega256D3 device.

[file #28839: xmegaD3-CRC-Power-corrections.patch](https://savannah.nongnu.org/bugs/download.php?file_id=28839)

This issue was migrated from https://savannah.nongnu.org/bugs/?39783

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Wed 14 Aug 2013 09:05:08 AM CEST

Attached patch "xmegaD3-CRC-Power-corrections.patch" fixes this issue.

ChangeLog
---------
2013-08-14  Pitchumani Sivanupandi <pitchumani.s@atmel.com>

* include/avr/iox64d3.h: Added CRC definitions. Corrected power reduction
definitions.
* include/avr/iox128d3.h: Same.
* include/avr/iox192d3.h: Same.
* include/avr/iox256d3.h: Added CRC and power reduction definitions.
* include/avr/power.h: Added power reduction macro definitions and docs.

sprintersb, 2024-02-07:
Fixed by 2d8308ea8c0ecc5a1e3913bfd63e86297d7f7621, 0c809eff79d96935ff33fc440f923849289c4b84.

closed 2024-02-07 by sprintersb