avrdudes/avr-libc #570: [bug #39875] Warning	1	unused variable '__ticks' [-Wunused-variable]	C:\PROGRA~2\Atmel\ATMELT~1\AVR8GC~1\Native\342~1.939\AVR8-G~1\av

avrs-admin, 2022-01-31:
Tue 27 Aug 2013 08:01:43 AM CEST

_delay_ms has the following definition:

void
_delay_ms(double __ms)
{
uint16_t __ticks;
double __tmp ;
#if **_HAS_DELAY_CYCLES && defined(__OPTIMIZE_**) && \
!defined(**_DELAY_BACKWARD_COMPATIBLE_**) &&	   \
 **_STDC_HOSTED_**
uint32_t __ticks_dc;
extern void __builtin_avr_delay_cycles(unsigned long);
__tmp = ((F_CPU) / 1e3) * __ms;

#if defined(**_DELAY_ROUND_DOWN_**)
__ticks_dc = (uint32_t)fabs(__tmp);

#elif defined(**_DELAY_ROUND_CLOSEST_**)
__ticks_dc = (uint32_t)(fabs(__tmp)+0.5);

#else
 [round](https://savannah.nongnu.org//round) up by default
__ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
#endif

__builtin_avr_delay_cycles(__ticks_dc);

#else
__tmp = ((F_CPU) / 4e3) * __ms;
if (__tmp < 1.0)
__ticks = 1;
else if (__tmp > 65535)
{
//	__ticks = requested delay in 1/10 ms
__ticks = (uint16_t) (__ms * 10.0);
while(__ticks)
{
// wait 1/10 ms
_delay_loop_2(((F_CPU) / 4e3) / 10);
__ticks --;
}
return;
}
else
__ticks = (uint16_t)__tmp;
_delay_loop_2(__ticks);
#endif
}

If it is possible for the first half of the conditional to be true (it was for me) then the definitions of _delay_ms and _delay_us are obviously incorrect.  (__ticks is not referenced before the #else clause.)  I suggest moving the declaration of __ticks to the #else clause, as follows:

void
_delay_ms(double __ms)
{
double __tmp ;
#if **_HAS_DELAY_CYCLES && defined(__OPTIMIZE_**) && \
!defined(**_DELAY_BACKWARD_COMPATIBLE_**) &&	   \
 **_STDC_HOSTED_**
uint32_t __ticks_dc;
extern void __builtin_avr_delay_cycles(unsigned long);
__tmp = ((F_CPU) / 1e3) * __ms;

#if defined(**_DELAY_ROUND_DOWN_**)
__ticks_dc = (uint32_t)fabs(__tmp);

#elif defined(**_DELAY_ROUND_CLOSEST_**)
__ticks_dc = (uint32_t)(fabs(__tmp)+0.5);

#else
 [round](https://savannah.nongnu.org//round) up by default
__ticks_dc = (uint32_t)(ceil(fabs(__tmp)));
#endif

__builtin_avr_delay_cycles(__ticks_dc);

#else
uint16_t __ticks;

__tmp = ((F_CPU) / 4e3) * __ms;
if (__tmp < 1.0)
__ticks = 1;
else if (__tmp > 65535)
{
//	__ticks = requested delay in 1/10 ms
__ticks = (uint16_t) (__ms * 10.0);
while(__ticks)
{
// wait 1/10 ms
_delay_loop_2(((F_CPU) / 4e3) / 10);
__ticks --;
}
return;
}
else
__ticks = (uint16_t)__tmp;
_delay_loop_2(__ticks);
#endif
}

This issue was migrated from https://savannah.nongnu.org/bugs/?39875

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Tue 27 Aug 2013 08:23:11 AM CEST

I think this is a duplicate for **[bug #38904](https://savannah.nongnu.org/bugs/?38904)**, even though the
warning message is different.

What do you think about it?

avrs-admin, 2022-01-31:
Mike Ady <farquahrsen>
Tue 27 Aug 2013 01:22:23 PM CEST

I did see #38904, and I thought that this was different.  In my experience, if there is any mystery about the "may be used uninitialized" error mentioned in #38904, then it is often caused by an unexpected interaction with a macro.  The OP might want to take a look at the preprocessor expansion in the listing file.

The fix that I suggested below does get rid of the "unused variable" warning message from the compiler.  I don't think that the change would have any effect on #38904.  It seems to me that the bug that I'm complaining about is a simple oversight and who ever introduced the bug failed to test both code paths of the preprocessor conditional.

sprintersb, 2024-04-12:
Fixed by 1b74e60f20e559bc9111e6ba1e5c07486389e5ed.

closed 2024-04-12 by sprintersb