avrdudes/avr-libc #571: [bug #40003] Integer type promotion leads to inefficent code in wdt.h

avrs-admin, 2022-01-31:
Wed 11 Sep 2013 05:15:33 PM CEST

Operands constructed in this fashion:
"r" (_BV(_WD_CHANGE_BIT) | _BV(WDE)),

...result in the use of two registers, one of which is loaded with 0x00 and never used.

Seems as though integer type promotion is responsible.  A fix is as simple as casting to an 8-bit type:
"r" ((uint8_t)(_BV(_WD_CHANGE_BIT) | _BV(WDE))),

Indeed, most **_asm_** statements in wdt.h are already taking this approach, but a couple are not.

Attached is a patch to resolve the issue.

I'm aware this is a minor issue, but as it is easy to resolve, I hope it will be accepted :)

Cheers,
JJ

[file #29105: wdt.h.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29105)

This issue was migrated from https://savannah.nongnu.org/bugs/?40003

avrs-admin, 2022-01-31:
Joey Morin <joeymorin>
Wed 11 Sep 2013 05:23:28 PM CEST

Forgot to add an example:

In the case of wdt_enable(WDTO_15MS), the assembled output for targets such as the ATtiny85:
ldi	r18, 0x08
ldi	r24, 0x18
ldi	r25, 0x00
in	r0, 0x3f
cli
wdr
out	0x21, r24
out	0x3f, r0
out	0x21, r18

With the patch in place:
ldi	r25, 0x08
ldi	r24, 0x18
in	r0, 0x3f
cli
wdr
out	0x21, r24
out	0x3f, r0
out	0x21, r25

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Mon 28 Apr 2014 03:40:15 PM CEST

committed the changes.
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2419](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2419)

sprintersb, 2024-02-03:
Fixed by 62fecd2511cbc9b637c83227fbd7f57655a674a1.

closed 2024-02-03 by sprintersb