avrdudes/avr-libc #574: [bug #40206] incorrect SP init in startup code for xmegas

avrs-admin, 2022-01-31:
Tue 08 Oct 2013 02:24:16 PM CEST

gcrt1.S writes SPH before SPL for all devices. For xmegas, the compiler, however, writes SPL before SPH. This is to avoid occurence of interrupts after the writing of SPH but before the writing of SPL; the hardware disables interrupts for four instructions after the write to SPL.

Also, things like writing to the CCP first up in main fail silently because of the reversed order of writing to the SP (when there are less than 4 instructions separating the write to SPL and the write to CCP).

The attached patch (**[patch #1](https://savannah.nongnu.org/patch/?1)**) makes the startup code write to SPL and then SPH for xmegas, making it consistent with what the compiler does for function prologues.

[file #29326: fix_sp_write_xmega.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29326)

This issue was migrated from https://savannah.nongnu.org/bugs/?40206

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Tue 29 Apr 2014 08:26:08 AM CEST

committed the changes.
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2421](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2421)

sprintersb, 2024-02-03:
Fixed by 5834ec42b4626f57b68c747025ee443fda30b545.

closed 2024-02-03 by sprintersb