avrdudes/avr-libc #576: [bug #40567] Invalid names in iotn13a.h (EEPE/EEMPE/BODS/BODSE)

avrs-admin, 2022-01-31:
Wed 13 Nov 2013 05:17:08 PM CET

As per subject, there are couple of wrong defines in the iotn13a.h file.

See also page 21 and 33 of [http://www.atmel.com/Images/doc8126.pdf](https://savannah.nongnu.orghttp://www.atmel.com/Images/doc8126.pdf)

Is fixing this directly ok? Or should the old (wrong) names be kept around for compatibility?

[file #29609: tn13a.patch](https://savannah.nongnu.org/bugs/download.php?file_id=29609)

This issue was migrated from https://savannah.nongnu.org/bugs/?40567

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Fri 21 Mar 2014 05:19:09 PM CET

fixed the header file
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2415](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2415)

avrs-admin, 2022-01-31:
Matthijs Kooijman <matthijs>
Sun 23 Mar 2014 09:07:25 PM CET

Thanks!

sprintersb, 2024-02-07:
Fixed by 5089e078a0165dd842f076b2e2397f9232a6a31b.

closed 2024-02-07 by sprintersb