avrdudes/avr-libc #579: [bug #41006] iom328p.h: wrong fuse defaults

avrs-admin, 2022-01-31:
Thu 26 Dec 2013 07:30:20 PM CET

The default fuse values defined in iom328p.h do not match the ATmega328(p) datasheet. Also, the header file documentation and placement of the defines suggest a wrong distribution of the fuse bits over the low, high and extended fuse bytes.

I have attached a patch and the resulting corrected header file.

[file #30100: iom328p-fixed.h](https://savannah.nongnu.org/bugs/download.php?file_id=30100)
[file #30099: iom328p.patch](https://savannah.nongnu.org/bugs/download.php?file_id=30099)

This issue was migrated from https://savannah.nongnu.org/bugs/?41006

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Fri 02 May 2014 10:30:24 AM CEST

HFUSE byte changes are already in header file. Incorporated the EFUSE_DEFAULT changes.

[http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2429](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2429)

This patch has been applied and committed.

Thank you for the contribution.

sprintersb, 2024-02-07:
Fixed by 30a2eddf943b2111cd8879b1c7d001eeca464671.

closed 2024-02-07 by sprintersb