avrdudes/avr-libc #586: [bug #42024] build break regarding avrtiny10

avrs-admin, 2022-01-31:
Tue 01 Apr 2014 10:47:49 PM CEST

I just pulled the avr-libc trunk and got this compilation error:

avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include  -I../../../common -I../../../include -I../../../include -x assembler-with-cpp  -mmcu=avrtiny10 -D__COMPILING_AVR_LIBC__   -MT utoa_ncheck.o -MD -MP -MF .deps/utoa_ncheck.Tpo -c -o utoa_ncheck.o ../../../libc/misc/utoa_ncheck.S
../../../libc/misc/utoa_ncheck.S: Assembler messages:
../../../libc/misc/utoa_ncheck.S:80: Error: illegal opcode sbiw for mcu avrtiny10
make[5]: *** [utoa_ncheck.o] Error 1
make[5]: Leaving directory `/home/baxelrod/lca/avr-libc/trunk/avr/lib/avrtiny10'
make[4]: *** [all-recursive] Error 1
make[4]: Leaving directory `/home/baxelrod/lca/avr-libc/trunk/avr/lib/avrtiny10'
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory `/home/baxelrod/lca/avr-libc/trunk/avr/lib'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/home/baxelrod/lca/avr-libc/trunk/avr'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/baxelrod/lca/avr-libc/trunk'
make: *** [all] Error 2

System configured like so:
./bootstrap
./configure --build=`./config.guess` --host=avr make On Ubuntu 12.04 with avr-gcc (GCC) 4.5.3.

After poking around the email archive, it looks like there have been other issues like this and the result was to not compile those libraries with protection like so:
#if !defined(**_AVR_TINY_**)

I looked into utoa_ncheck.S and found that it had this:
#if !defined(__AVR_TINY).
Is it possible that the protection variable changed names at some point?  I found several other files that had this (presumably) improper protection variable:

./libc/misc/utoa_ncheck.S
./libc/misc/ultoa.S
./libc/misc/ultoa_ncheck.S
./libc/misc/itoa.S
./libc/misc/ltoa_ncheck.S
./libc/misc/ltoa.S
./libc/misc/utoa.S
./libc/misc/itoa_ncheck.S

I fixed them and provided it in the attached patch.

NOTE: patch is untested other than compiling entire library.  I don't have a avrtiny to test with.

[file #31097: tinypatch.diff](https://savannah.nongnu.org/bugs/download.php?file_id=31097)

This issue was migrated from https://savannah.nongnu.org/bugs/?42024

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Mon 28 Apr 2014 04:06:04 PM CEST

committed the changes.
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2420](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2420)

sprintersb, 2024-02-07:
Fixed by 83c79045acbd412d95b9913d54833aad44dda620.

closed 2024-02-07 by sprintersb