avrdudes/avr-libc #587: [bug #42084] wrong LFUSE_DEFAULT in iotn84a.h

avrs-admin, 2022-01-31:
Thu 10 Apr 2014 03:11:23 AM CEST

#define LFUSE_DEFAULT (FUSE_CKSEL0 & FUSE_CKSEL2 & FUSE_SUT0 & FUSE_CKDIV8)
doesn't match datasheet table 19-5.

It should be the same as iotn84.h:
#define LFUSE_DEFAULT (FUSE_CKSEL0 & FUSE_CKSEL2 & FUSE_CKSEL3 & FUSE_SUT0 & FUSE_CKDIV8)

This issue was migrated from https://savannah.nongnu.org/bugs/?42084

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Wed 16 Apr 2014 01:45:03 PM CEST

Updated LFUSE_DEFAULT in iotn84a.h as per datasheet.
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2418](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2418)

sprintersb, 2024-02-07:
Fixed by 45ea6b5d7b508f38313afb4016566d4851ac9d35.

closed 2024-02-07 by sprintersb