avrdudes/avr-libc #593: [bug #43044] memory sections require __attribute__ ((used))

avrs-admin, 2022-01-31:
Thu 21 Aug 2014 09:36:23 PM CEST

Memory sections does not always work without used attribute.  Docs should reflect this.
 [http://www.nongnu.org/avr-libc/user-manual/mem_sections.html](https://savannah.nongnu.orghttp://www.nongnu.org/avr-libc/user-manual/mem_sections.html)

[http://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=1180180](https://savannah.nongnu.orghttp://www.avrfreaks.net/index.php?name=PNphpBB2&file=viewtopic&p=1180180)

[file #33360: doc.patch](https://savannah.nongnu.org/bugs/download.php?file_id=33360)
[file #33361: doc.patch](https://savannah.nongnu.org/bugs/download.php?file_id=33361)

This issue was migrated from https://savannah.nongnu.org/bugs/?43044

avrs-admin, 2022-01-31:
Senthil Kumar Selvaraj <saaadhu>
Wed 18 Mar 2015 07:48:45 AM CET

Committed in rev 2471 - [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2471](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2471)

sprintersb, 2024-02-03:
Fixed by ba7a97364b1e48d4e2b8a317aad65c937bcd3a9a.

closed 2024-02-03 by sprintersb