avrdudes/avr-libc #594: [bug #43049] avr-libc 1.8.1 cannot be built with --enable-debug-info=dwarf-4

avrs-admin, 2022-01-31:
Fri 22 Aug 2014 10:37:27 AM CEST

Trying to build avr-libc 1.8.1 with version the "--enable-debug-info=dwarf-4" configure option does not work, as it results in an error complaining about pgm_copystring().

avr-gcc -DHAVE_CONFIG_H -I. -I../../..  -I../../../common -I../../../include -I../../../include -D_FORTIFY_SOURCE=2 -gdwarf-4 -Wall -W -Wstrict-prototypes -mmcu=avr2 -D__COMPILING_AVR_LIBC__ -mcall-prologues -Os  -MT strftime.o -MD -MP -MF .deps/strftime.Tpo -c -o strftime.o ../../../libc/time/strftime.c
../../../libc/time/strftime.c: In function 'pgm_copystring':
../../../libc/time/strftime.c:56:1: internal compiler error: in convert_debug_memory_address, at cfgexpand.c:3547
pgm_copystring(const char __memx * p, unsigned char i, char *b, unsigned char l)
^
Please submit a full bug report,
with preprocessed source if appropriate.
See <[http://gcc.gnu.org/bugs.html](https://savannah.nongnu.orghttp://gcc.gnu.org/bugs.html)> for instructions.

This was initially reported here [1], I just want to bring this to the attention of all avr-libc developers.

Furthermore I would like to add the fact that version 1.8.1 cannot be chosen as Release in this bug tracker, so I had to choose "Any". Probably someone simply forgot to add the release into the database ;).

[1]: [https://www.mikrocontroller.net/topic/341504?goto=new#3769337](https://savannah.nongnu.orghttps://www.mikrocontroller.net/topic/341504?goto=new#3769337)

This issue was migrated from https://savannah.nongnu.org/bugs/?43049

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Fri 22 Aug 2014 10:56:09 AM CEST

Sorry, but this is a plain GCC bug.  There's nothing avr-libc
could do about it.

I already filed the respective GCC bug some days ago:

[https://gcc.gnu.org/bugzilla/show_bug.cgi?id=62084](https://savannah.nongnu.orghttps://gcc.gnu.org/bugzilla/show_bug.cgi?id=62084)

avrs-admin, 2022-01-31:
-X- <johnpatcher>
Fri 22 Aug 2014 11:05:48 AM CEST

Alright, thanks. Just wanted to make sure this gets some attention from the correct people (since you haven't responded in the mikrocontroller.net thread ;)). Not a show-stopper for me, but definitely something that should be fixed at some point in the future.

sprintersb, 2025-07-25:
GCC issue.

closed 2025-07-25 by sprintersb