avrdudes/avr-libc #597: [bug #43261] symbols in documentation are getting their leading and trailing "__" stripped by doxygen

avrs-admin, 2022-01-31:
Fri 19 Sep 2014 07:10:50 PM CEST

For example this page:

 [http://www.nongnu.org/avr-libc/user-manual/using_tools.html](https://savannah.nongnu.orghttp://www.nongnu.org/avr-libc/user-manual/using_tools.html)

list all the macros like AVR_ARCH instead of **_AVR_ARCH_**

Joerg says thi sis a doxygen issue and points out that in the
next table a few lines below the underscores are working fine:

Strange enough, in the next table a few lines below, the
underscores are working fine:

<tr><td>avr1</td><td>at90s1200</td<td>__AVR_AT90S1200__</td></tr>

This issue was migrated from https://savannah.nongnu.org/bugs/?43261

avrs-admin, 2022-01-31:
Jonathan W. Taylor <elderling>
Sun 14 Aug 2016 11:05:52 PM CEST

Generating the documentation with doxygen 1.8.11 appears to resolve this issue completely.

avrs-admin, 2022-01-31:
Jonathan W. Taylor <elderling>
Mon 15 Aug 2016 12:29:37 AM CEST

I can see the CVS repo where the documentation is stored, but I don't know whether it is acceptable for me to change it. Furthermore, if it OK for me to change it, I don't know if I should regenerate all of the documentation or just the broken files.

sprintersb, 2024-03-24:
Semms to be one of the Doxygen glitches.  With Doxygen v1.9.1 (or v1.8.11 according to comment above) the underscores are there, so a fix is to re-generate the documentation. See also #950.

sprintersb, 2024-03-24:
Dup of #950.

closed 2024-03-24 by sprintersb