avrdudes/avr-libc #606: [bug #44514] Wrong build option '-mtiny-stack' instead of '-msp8'

avrs-admin, 2022-01-31:
Wed 11 Mar 2015 08:37:20 PM CET

Libraries are being built with a wrong option -mtiny-stack:

devtools/gen-avr-lib-tree.sh:49

```
CFLAGS_TINY_STACK="-mtiny-stack -mcall-prologues -Os"

```

To query avr-gcc for it's multilib layout, use -print-multi-lib:

```
$ avr-gcc -print-multi-lib
.;
avr25;@mmcu=avr25
avr3;@mmcu=avr3
avr31;@mmcu=avr31
avr35;@mmcu=avr35
avr4;@mmcu=avr4
avr5;@mmcu=avr5
avr51;@mmcu=avr51
avr6;@mmcu=avr6
avrxmega2;@mmcu=avrxmega2
avrxmega4;@mmcu=avrxmega4
avrxmega5;@mmcu=avrxmega5
avrxmega6;@mmcu=avrxmega6
avrxmega7;@mmcu=avrxmega7
tiny-stack;@msp8
avr25/tiny-stack;@mmcu=avr25@msp8

```
For each line:  Left of ';' stands the multilib subdir, right of ';' are the option(s) to select that multilib subdir (replace '@' with ' -' to get the option(s)).

Notice that -mtiny-stack is **not** an avr-gcc multilib option!  The correct option is -msp8.

In order to query for the multilib subdir for a specific set of command options, use

```
$ avr-gcc -print-multi-directory <options>

```
Example:

```
$ avr-gcc -print-multilib -mmcu=avr2 -msp8 -Os
tiny-stack

```

'-Os' is ignored (no multilib option)
'-mmcu=avr2' is the default multilib, hence will yield '.'.
'-msp8' selects the tiny-stack subdir for avr2 and the result is './tiny-stack' i.e. 'tiny-stack'.

Also notice that apart from their multilib properties, -msp8 and -mtiny-stack are working differently:

- -msp8 asserts that the stack pointer (SP) is 8 bits wide **physically**.  avr-gcc will set -msp8 as needed except for the cases -mmcu=avr2 and -mmcu=avr25.  These are the only core architectures which intermix devices with 16-bit wide and 8-bit wide SP.  Or, to put it more precisely:  avr-gcc uses -msp8 to decomposes these to sets of devices into 4 core architectures.- -mtiny-stack is an optimization option:  Only the lower 8 bits of SP will be changed, no matter how SP is layed out physically.  This applies to all devices and archirectures.

[file #33357: flags.patch](https://savannah.nongnu.org/bugs/download.php?file_id=33357)

This issue was migrated from https://savannah.nongnu.org/bugs/?44514

avrs-admin, 2022-01-31:
Senthil Kumar Selvaraj <saaadhu>
Wed 18 Mar 2015 07:55:04 AM CET

Committed in rev 2472- [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2472](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2472)

sprintersb, 2024-02-03:
Fixed by 73ef1f7461007971f97e1debf99e571e9ebc9500.

closed 2024-02-03 by sprintersb