avrdudes/avr-libc #609: [bug #44574] Place device-specific libraries in their multilib directory.

avrs-admin, 2022-01-31:
Wed 18 Mar 2015 08:10:31 PM CET

The current naming scheme for device-specific libraries

```
dev/<DEVICE>/libdev.a

```
bears the problem that it won't work if the install path contains spaces (as not uncommon under MS windows).

This is actually not an avr-libc problem but a problem with avr-gcc.  However, the new device-specific spec files have very limited means to handle paths containing spaces.

Wrapping the path in a space-escaping spec function does not work because that would require more than one level of escaping (which is not possible because the escape character would become a part of the path name).

One approach to make avr-gcc 5.0 + avr-libc work with paths containing spaces is renaming the device library to

```
lib<DEVICE>.a

```
and installing it in its multilib directory, i.e. in the multilib subdirectory as reported by

```
$ avr-gcc -mmcu=<DEVICE> [options] -print-multi-directory

```
For devices not natively supported by avr-gcc, avr-libc would have to know the correct multilib subdir.

Rationale is that multilib subdirs are common library search paths.  Hence the device library could then be specified as

```
-l<DEVICE>

```
and thus without explicit absolute path which works as expected.

[file #33665: move-crt-lib-to-multilib-location.patch](https://savannah.nongnu.org/bugs/download.php?file_id=33665)

This issue was migrated from https://savannah.nongnu.org/bugs/?44574

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Mon 13 Apr 2015 02:47:36 PM CEST

Attached patch to changes the startup object and device specific libraries name.
Also place them into multilib (architecture) location.
Startup object file name: crt<device>.o
Device specific lib name: lib<device>.a

Old startup object file name (e.g. crtm1280.o) is not removed from the device list yet.
This may be required when we make avr-libc compile-able with older gcc version.

ChangeLog

```
2015-04-13  Pitchumani Sivanupandi <pitchumani.s@atmel.com>

bug #44574: Place device-specific libraries in their multilib directory.
* configure.ac (AC_CONFIG_FILES): Replace dev directory with architecture
directory where device specific lib (lib<device>.a) and startup object
file (crt<device>.o) to be installed.
* devtools/Devices.am (avr_LIBRARIES): Change device library name to
lib<device>.a.
* devtools/gen-avr-lib-tree.sh (AVR12_DEV_INFO): Remove tiny stack devices
as they are part of AVR12TS_DEV_INFO.
(AVR25_DEV_INFO): Ditto.
Change startup object file name to crt<device>.o and install location to
multilib (architecture) directory.

```

(**[file #33665](https://savannah.nongnu.org/support/download.php?file_id=33665)**)

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Wed 15 Apr 2015 12:00:21 PM CEST

Committed.
 [http://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2474](https://savannah.nongnu.orghttp://svn.savannah.nongnu.org/viewvc?view=rev&root=avr-libc&revision=2474)

sprintersb, 2024-02-04:
Fixed by 4b4d889093544d27816f0a4150ec9b8ccb4a8c17.

closed 2024-02-04 by sprintersb