avrdudes/avr-libc #617: [bug #45151] #define _ASM_SFR_COMPAT 1 and #include <avr/power.h> results in build error

avrs-admin, 2022-01-31:
Wed 20 May 2015 04:49:05 PM CEST

Note: release field is incorrect, but there isnt one for newest svn.
I was testing current trunk avr-libc + current gcc 5 branch
(20.5.2015, avr-libc r2475) against my projects when i found this.

I simplified it as:

$ cat simple.c
#define _SFR_ASM_COMPAT 1
#include <avr/io.h>
#include <avr/power.h>

void main(void) { }

$ avr-gcc -mmcu=atmega328p -Os simple.c
In file included from simple.c:3:0:
/home/urjaman/avrtc-test/avr/include/avr/power.h: In function '__power_all_enable':
/home/urjaman/avrtc-test/avr/include/avr/power.h:1152:9: error: lvalue required as left operand of assignment
PRR &= (uint8_t)~(__AVR_HAVE_PRR);
^
/home/urjaman/avrtc-test/avr/include/avr/power.h: In function '__power_all_disable':
/home/urjaman/avrtc-test/avr/include/avr/power.h:1201:9: error: lvalue required as left operand of assignment
PRR |= (uint8_t)(__AVR_HAVE_PRR);
^

I think this might have been officially not supported, but it used to work without error.

I worked around by not including <avr/power.h> in the module that uses _ASM_SFR_COMPAT (has a few variable definitions and an ASM ISR and some C helpers to interface with the variables (and thus the ISR)), but it is slightly annoying for a project where all system includes used to in one main.h (another .c needed to include <avr/power.h> then).

This issue was migrated from https://savannah.nongnu.org/bugs/?45151

avrs-admin, 2022-01-31:
Pitchumani <pitchumani>
Thu 21 May 2015 03:31:16 PM CEST

If _SFR_ASM_COMPAT is defined, you can't use any of the macros/ functions from power.h. Now __power_all_enable/disable are modified as inline function, so those functions are expanded where power.h is included and caused these errors.

Possible solution maybe to include power.h conditionally (if not _SFR_ASM_COMPAT)

sprintersb, 2024-04-21:
Closed, this is not supposed to work together.

closed 2024-04-21 by sprintersb