avrdudes/avr-libc #620: [bug #45304] Will it be possible please to add support and document all possible interrupts for ATmega16/32HVB & ATmega8/16HVA

avrs-admin, 2022-01-31:
Thu 11 Jun 2015 07:13:27 PM CEST

Will it be possible please to add support and document all possible interrupts
for ATmega16/32HVB & ATmega8/16HVA

from
 [http://www.atmel.com/webdoc/AVRLibcReferenceManual/group__avr__interrupts.html](https://savannah.nongnu.orghttp://www.atmel.com/webdoc/AVRLibcReferenceManual/group__avr__interrupts.html)
run

#include <avr/interrupt.h>
ISR(ADC_vect)
{
// user code here
}

an an ATmega32HVB in atmel studio 6
start adc conversion on cell 1 to 4
the interrupt will not be triggered, and the avrlib reference manual does not
specify any other interrupt besides for ADC conversion, also trying to compile
the code generates an warning.

<i tried all this in an attempt to find if any of this are defined>
Warning    1    'ADC_vect' appears to be a misspelled signal handler
Warning    4    'ANALOG_COMP_0_vect' appears to be a misspelled signal handler
Warning    5    'ANALOG_COMP_1_vect' appears to be a misspelled signal handler
Warning    6    'ANALOG_COMP_2_vect' appears to be a misspelled signal handler
Warning    7    'ANALOG_COMP_vect' appears to be a misspelled signal handler
Warning    8    'ANA_COMP_vect' appears to be a misspelled signal handler

now there is clearly an interrupt there because the default int is being
triggered on VADCSR |= 1<<VADSC;
ISR(__vector_default)

also CADIC_vect interrupt is missing for any cpu (this is the IAR name but
could not find any equivalent in the avr-gcc even for other cpu)

also if we go to the avr-lib doc (the above link) and we search for atmega32hvb
we will find that it supports just some generic  interrupts like external/timer
.. nothing more witch makes me believe the support is missing ffor this micro
controller..

in conclusion either some interrupt names are missing from the definition files
for this micro or either the documentation is incomplete (see the above link
used for reference).

Will it be possible please to add support and document all possible interrupts
for ATmega16/32HVB & ATmega8/16HVA

This issue was migrated from https://savannah.nongnu.org/bugs/?45304

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Thu 11 Jun 2015 09:42:05 PM CEST

All the interrupt vectors are there:

```
$ fgrep _vect ~/src/avr-libc/include/avr/iom32hvb.h | fgrep -v _vect_num | awk '{print $2}'
BPINT_vect
VREGMON_vect
INT0_vect
INT1_vect
INT2_vect
INT3_vect
PCINT0_vect
PCINT1_vect
WDT_vect
BGSCD_vect
CHDET_vect
TIMER1_IC_vect
TIMER1_COMPA_vect
TIMER1_COMPB_vect
TIMER1_OVF_vect
TIMER0_IC_vect
TIMER0_COMPA_vect
TIMER0_COMPB_vect
TIMER0_OVF_vect
TWIBUSCD_vect
TWI_vect
SPI_STC_vect
VADC_vect
CCADC_CONV_vect
CCADC_REG_CUR_vect
CCADC_ACC_vect
EE_READY_vect
SPM_vect

```
It's our general policy to follow the exact (*) naming of
the datasheet with our naming.  So, the only issue here is
that these HVA/HVB devices use an interrupt naming that
vastely differs from many other AVRs.

(*) As exact as possible in terms of a valid C name, of
course, so the datasheets "SPI, STC" is converted to
"SPI_STC" for the C name, for example.

The documentation used to be auto-generated from the old
XML files.  Alas, that process has never been moved forward
to the new XML, so yes, documentation might be missing.  I'm
leaving the bug open for that.

sprintersb, 2024-03-24:
IMO it's not feasible to document all the ISR names. Taken over all devices, we currently have around 12000 (twelve thousand) IRQ vectors. To make thinks worse, same IRQ purpose may have different namings on different MCUs; and it's not feasible to support the header files by hand, either...

sprintersb, 2024-05-23:
Fixed by 5021ae0eb77626a4969c253c10f53da52e522177.

closed 2024-05-23 by sprintersb