avrdudes/avr-libc #625: [bug #45701] redundant stack initialization in crt1.o

avrs-admin, 2022-01-31:
Wed 05 Aug 2015 06:03:13 PM CEST

It appears crt1.o always initializes the stack and clears SREG, even for modern AVRs that don't require it.
While it is necessary for old AVRs like the mega8, since SPH/SPL is initialized to 0 on reset (datasheet s. 7.5.1), on the tinyx5, tinyx4, mega48/88/168/328, and many others SPH/SPL is initialized to RAMEND upon reset.  See, for example, s. 4.8.1 of the tiny84a datasheet.

This issue was migrated from https://savannah.nongnu.org/bugs/?45701

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Mon 11 Jul 2016 09:16:05 PM CEST

The current startup code allows a custom value for SP init by defining symbol  __stack , for example:  -Wl,--defsym,__stack=0x123  or by defining that symbol in a custom linker description file.

Moreover it is not uncommon to (re)start an application by jumping to 0x0, e.g. from a bootloader.  If the startup code was "optimized" as proposed, that code will no more work as expected.

avrs-admin, 2022-01-31:
Mon 18 Jul 2016 09:14:15 PM CEST

Acutally a bootloader just jumping to 0x0 would not only be uncommon, it would be shitty coding.  All the bootloaders I have written ensure the registers are in their reset state before jumping to the start of the application code.  Another option is to use a WDT reset like Optiboot.

sprintersb, 2024-02-03:
The current implementation allows to initialize SP to the value of symbol `__stack`. Dropping that would remove the feature.

closed 2024-02-03 by sprintersb