avrdudes/avr-libc #634: [bug #48473] FAQ: "How do I relocate code to a fixed address?" can be misleading

avrs-admin, 2022-01-31:
Mon 11 Jul 2016 08:26:31 PM CEST

The method described in [http://www.nongnu.org/avr-libc/user-manual/FAQ.html](https://savannah.nongnu.orghttp://www.nongnu.org/avr-libc/user-manual/FAQ.html) > "How do I relocate code to a fixed address?" does not work in every situation.

A function can get inlined during optimization. In this case the code is not run from the specified address as I would expect.

I recently had a problem writing the FLASH on XMEGAs because of this (as the SPM instruction has to reside in the boot section of FLASH). It took me a while to debug this. My solution was to use the following attribute:

**_attribute_** ((noinline, section (".boot")))

I suggest putting a note about this case. I think it would be very useful for others like me.

[file #37813: inline_section_bug_sample.zip](https://savannah.nongnu.org/bugs/download.php?file_id=37813)

This issue was migrated from https://savannah.nongnu.org/bugs/?48473

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Mon 11 Jul 2016 08:44:59 PM CEST

If inlining is a problem, then the same applies to cloning.  Hence the docs should also recommend  __noclone__  and  __no_icf__  (provided gcc supports these attributes).

sprintersb, 2024-04-22:
Added as https://github.com/avrdudes/avr-libc/commit/d5adaf57f316fdcd2079a34faded46e145f2946d.

closed 2024-04-22 by sprintersb