avrdudes/avr-libc #636: [bug #48765] math.h: isfinite might generate wrong code due to early-clobber

avrs-admin, 2022-01-31:
Fri 12 Aug 2016 06:59:39 PM CEST

Skimming math.h I came across the following implementation of isfinie:

```
__ATTR_CONST__ static inline int isfinite (double __x)
{
unsigned char __exp;
__asm__ (
"mov        %0, %C1                \n\t"
"lsl        %0                \n\t"
"mov        %0, %D1                \n\t"
"rol        %0                "
: "=r" (__exp)
: "r" (__x)        );
return __exp != 0xff;
}

```
The potential problem is that nothing keeps the compiler from assigning the registers in such a way that %0 = %D1, i.e. there might be an earaly-clobber situation where the first MOV instruction overwrites %D1 with %C1.

Tried around but I could not trick avr-gcc into generating wrong code, but the problem is obviously at hand.

A possible fix is to use tmp_reg in the first two instructions:

```
__ATTR_CONST__ static inline int isfinite (double __x)
{
unsigned char __exp;
__asm__ (
"mov __tmp_reg__, %C1" "\n\t"
"lsl __tmp_reg__"      "\n\t"
"mov %0, %D1"          "\n\t"
"rol %0"
: "=r" (__exp)
: "r" (__x)        );
return __exp != 0xff;
}

```
This avoids a "=&r" early-clobber of the output operand.  It works because the outcome of the first 2 instructions is only to set carry.

This issue was migrated from https://savannah.nongnu.org/bugs/?48765

avrs-admin, 2022-01-31:
Chris Pavlina <c4757p>
Fri 12 Aug 2016 09:51:57 PM CEST

I'm not so sure. %C1 and %D1 are defined as bytes of __x, and %0 is __exp - surely the compiler should keep those separate as they are separate variables existing in the same scope?

I'm going to mark this Need Info unless someone else can correct me on this. I can't see the potential issue here.

The only way I was able to force a collision was to rewrite my own version of isfinite() with 'register unsigned char __exp asm("r27")' to force it to use a colliding register. Otherwise, surely it should be required to keep them separate while they are still valid in the scope?

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Fri 12 Aug 2016 10:34:58 PM CEST

No, that's not how GCC inline assembler works.  The only constraints on the registers are:

o input: must be in a GRP register and starts with an even register number (as it's wider than 1 byte, i.e. %C1 is even, %D1 is odd).

o output: must be in a GRP (and it's completely fine if it overlaps the input).

Suppose you want to swap bytes of a 2-byte value as follows:

```
int swap (int x)
{
int y;
__asm ("mov %A0, %B1"  "\n\t"
"mov %B0, %A1"
: "=r" (y) : "r" (x));
return y;
}

```
$ avr-gcc -S -Os file.c

will get you:

```
swap:
/* #APP */
mov r24, r25
mov r25, r24
/* #NOAPP */
ret

```
Bummer! (and GCC is right)

This is because the input dies in the asm and is no more needed, and GCC takes the asm as being one chunk of code executed at the SAME time, not as a sequence if instructions.

If there is such an early-clobber situation, i.e. bits of the input are needed after the asm started writing the outputs, you'll need special care like early-clobber '&':

```
int swap (int x)
{
int y;
__asm ("mov %A0, %B1"  "\n\t"
"mov %B0, %A1"
: "=&r" (y) : "r" (x));
return y;
}

```
This yields correct

-verbatim-
swap:
mov r18,r24
mov r19,r25
/* #APP */
mov r24, r19
mov r25, r18
/* #NOAPP */
ret
-verbatim-

Notice that if you swapped nibbles of a 1-byte value by a single SWAP instruction, it's completely fine if input and output live in the same register, no early-clobber is needed.

GCC history tells us (and also Mr. Murphy) that if there is the potential for a bug, the compiler will take that route, be it sooner or later...

avrs-admin, 2022-01-31:
Chris Pavlina <c4757p>
Fri 12 Aug 2016 10:41:02 PM CEST

Okay, your example proves me wrong. Yes, this is indeed a potential bug.

The proposed replacement looks good to me.

Thank you for your report.

sprintersb, 2024-02-07:
Fixed by 556b4a8b3fee2be347c9f40f05a9cb2d891ea78b.

closed 2024-02-07 by sprintersb