avrdudes/avr-libc #639: [bug #49150] Incorrect bit definitions in io90pwm3b.h

avrs-admin, 2022-01-31:
Wed 21 Sep 2016 06:04:56 PM CEST

Two incorrect bit definitions in the General Timer/Counter Control Register GTCCR:

According to the Atmel datasheet (document 4317, section 13.4), ICPSEL1 is bit 6, and TSM is bit 7.

According to io90pwm3b.h, lines 261 & 262, those bits are 2 and 3 respectively.

I have confirmed that ICPSEL1 needs to be 6 by testing on actual hardware.

The incorrect definitions are present in version 2.0 and have been present since the file was added.  Odd that it hasn't been noticed before, although it does require special circumstances:  user must be doing captures with timer 1, and must be using capture input B, pin PB6.  (The default is to use input A, pin PD4.)

This issue was migrated from https://savannah.nongnu.org/bugs/?49150

avrs-admin, 2022-01-31:
Chris Pavlina <c4757p>
Thu 22 Sep 2016 12:49:56 AM CEST

Yes, it appears that these are incorrect. Thank you for your report.

sprintersb, 2024-03-16:
Fixed by https://github.com/avrdudes/avr-libc/commit/fdb9ca8fe2d2552fdeeec925e9901cfe766fffec.

closed 2024-03-16 by sprintersb