avrdudes/avr-libc #647: [bug #49984] fabs missing from libm implementation.

avrs-admin, 2022-01-31:
Wed 04 Jan 2017 12:19:01 PM CET

fabs is missing from the libm implementation:

```
#include <math.h>

float use_fabsf (float x)
{
return fabsf (x);
}

double use_fabs (double x)
{
return fabs (x);
}

int main (void) { return 0; }

```
```

```
```
__ATTR_CONST__ static inline int isfinite (double __x)
^~~
/srv/local/gnu/install/gcc-6/avr/include/math.h:359:42: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'double'
__ATTR_CONST__ static inline double copysign (double __x, double __y)

```

This issue was migrated from https://savannah.nongnu.org/bugs/?49984

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Wed 04 Jan 2017 12:21:21 PM CET

Some text has been thrown away by the web interface:

- Linker error re. missing fabs implementation occurs of compiled with -ffreestanding or with -fno-builtin- Compiler error because of unknown key word "inline" occurrs in C90 mode.

sprintersb, 2024-02-07:
Fixed by e08bbd25e260804090e3e55e40bbcac3dc8f4abd.

closed 2024-02-07 by sprintersb