avrdudes/avr-libc #652: [bug #50501] eeprom_update_byte() can reduce wear more

avrs-admin, 2022-01-31:
Thu 09 Mar 2017 06:04:53 AM CET

If the new value does not contain any 1 bits where the old value contains a 0 bit (ANDing the new value with the old value yields the new value), the cell does not have to be erased and can simply be overwritten with the new value.

This issue was migrated from https://savannah.nongnu.org/bugs/?50501

avrs-admin, 2022-01-31:
Eric Tang <e_l_tang>
Thu 09 Mar 2017 06:12:53 AM CET

This test should be performed in addition to the current test for equality, not in lieu of it.

avrs-admin, 2022-01-31:
Jan Waclawek <wek>
Thu 09 Mar 2017 09:12:19 AM CET

I wouldn't recommend to do this unless explicitly allowed by datasheet or an Atmel insider approves it. Depending on the particular implementation of EEPROM, write distrubs may decrease read reliability. This is not something which can be qualified by "I tried and it worked for me".

sprintersb, 2024-03-18:
See also #555

closed 2025-07-25 by sprintersb