avrdudes/avr-libc #655: [bug #50987] Not all registers need to be declared volatile

avrs-admin, 2022-01-31:
Tue 09 May 2017 10:20:25 PM CEST

Many registers are no different from a global 8-bit variables in that they will always contain the values which were last written to them. The compiler will be able to better optimize code which accesses them if they are not declared volatile.

This issue was migrated from https://savannah.nongnu.org/bugs/?50987

avrs-admin, 2022-01-31:
Georg-Johann Lay <gjlayde>
Tue 09 May 2017 10:33:26 PM CEST

Would you be a /bit/ more specific?

avrs-admin, 2022-01-31:
Ambroz Bizjak <abizjak>
Tue 09 May 2017 10:51:41 PM CEST

I think it is generally not a good idea to remove volatile from registers because writing/reading registers generally still has a side effect in addition to just changing the value.

For example, with volatile register accesses are never reordered (a different order may have different semantics).

Another example is that some infinite loops like while(1){REG=1;} would become undefined behavior (at least in C++) without REG being volatile. See: [http://en.cppreference.com/w/cpp/language/memory_model](https://savannah.nongnu.orghttp://en.cppreference.com/w/cpp/language/memory_model) (Progress guarantee).

In summary I think the only case where removing volatile from a register is where it has no specific hardware semantics at all - reading or writing has no side effect and the value itself does not influence anything.

avrs-admin, 2022-01-31:
Sat 23 Dec 2017 11:33:22 AM CET

As Ambroz explained, removing volatile is only safe if there are no (side) effects. This case IMHO never happens (each register is there for a specific purpose/effect). The registers with the least hardware side effects probably are the GPIOn registers, which are effectively like global variables. When I use them, it's usually in ISRs, thus, they must be declared volatile.

Voting to close the bug as wontfix.

sprintersb, 2024-02-07:
Closed accordnig to the comments. Plus, it's almost impossible here to keep and maintain a list of which registere behave that way, and to integrate that into device headers.  Supporting this would just be a nightmare.

closed 2024-02-07 by sprintersb