avrdudes/avr-libc #662: [bug #54652] Missing support for ATmega328PB

avrs-admin, 2022-01-31:
Wed 12 Sep 2018 07:29:13 PM CEST

(copied from [https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=867235](https://savannah.nongnu.orghttps://bugs.debian.org/cgi-bin/bugreport.cgi?bug=867235))

Attempting to build a program for the new ATmega328PB, it seems the device-specific IO file is not found:

avr-gcc -std=gnu99 -Wall -Os -DF_CPU=16000000 -mmcu=atmega328pb -flto -ffunction-sections ...
In file included from src/test.c:1:0:
/usr/lib/avr/include/avr/io.h:623:6: warning: #warning "device type not defined" [-Wcpp]
#    warning "device type not defined"
^

I do have the required iom328pb.h file though:

-rw-r--r-- 1 root root 27K Jul 22  2016 /usr/lib/avr/include/avr/iom328pb.h

Further, I see that iom328pb.h isn't listed in the big long section of #ifdef-guarded #includes in the main io.h. I see there is an attempted generic fallback section based on value of **_AVR_DEV_LIB_NAME_** but it seems for whatever reason that isn't kicking in today.

If I simply add the required 2 lines (by copying the 328P example), my code will compile fine. Attached is my patch and the new file.

[file #44993: add-iom328pb-h.diff](https://savannah.nongnu.org/bugs/download.php?file_id=44993)
[file #44992: iom328pb.h](https://savannah.nongnu.org/bugs/download.php?file_id=44992)
[file #45510: add_m328pb_to_svn_r2548.diff](https://savannah.nongnu.org/bugs/download.php?file_id=45510)

This issue was migrated from https://savannah.nongnu.org/bugs/?54652

avrs-admin, 2022-01-31:
Thu 22 Nov 2018 04:31:33 AM CET

Thanks Paul for getting this started.

I've attached a new patch (add_m328pb_to_svn_r2548.diff) that can be applied to the now-current (rev 2548) svn trunk and which will add support for the atmega328pb.  This includes edits to various makefiles and scripts so that support gets built in, plus two critical files: (1) atmega328pb.S which I generated using the existing devtools/ioreg.pl against the ATmega328PB.atdf XML file provided by Atmel in the now-current ATmega_DFP.1.2.272.atpack that I retrieved from [http://packs.download.atmel.com/](https://savannah.nongnu.orghttp://packs.download.atmel.com/); and (2) iom328pb.h which I pulled from this atpack as well.  One therefore need not use the Atmel-distributed binary copies of crtatmega328pb.o and libatmega328pb.a

It seems to work for me, but better support for this chip can/should be added, e.g. util/twi.h assumes only a single TWI module exists.

(**[file #45510](https://savannah.nongnu.org/support/download.php?file_id=45510)**)

avrs-admin, 2022-01-31:
Paul Evans <leonerd>
Wed 05 Jun 2019 05:49:31 PM CEST

Any update on this? I'm keen to not have to regularly make that 2-line edit to my "io.h" file every time the debian package gets updated.

Is there a process by which this can make its way into a release?

avrs-admin, 2022-01-31:
Clews <clewsy>
Wed 16 Sep 2020 07:32:36 AM CEST

I'm also seeking and update please.

I think the attached iom328pb.h needs the TCCR3B register definitions to be revised:

#define TCCR3B  _SFR_MEM8(0x91)
#define CS30    0
#define CS31    1
#define CS32    2
#define WGM32   3
#define WGM33   4
#define ICES3   6
#define ICNC3   7

sprintersb, 2024-03-15:
Dup of #879.  The support added in #879 has `TCCR3B` according to the comment above.

closed 2024-03-15 by sprintersb