avrdudes/avr-libc #664: [bug #54977] ATtiny4313 missing PCMSK0

avrs-admin, 2022-01-31:
Wed 07 Nov 2018 12:04:29 AM CET

The data sheet defines registers PCMSK0, PCMSK1 and PCMSK2

iotn2313a.h correctly provides all three (with PCMSK being another name for PCMSK0):

#define PCMSK _SFR_IO8(0x020)
#define PCMSK0 _SFR_IO8(0x020)

iotn4313.h is missing PCMSK0, and has only PCMSK:

#define PCMSK _SFR_IO8(0x020)

The PCMSK0 definition ought to be added

[file #50447: 0001-Added-missing-PCMSK0-macro-for-ATtiny4313.patch](https://savannah.nongnu.org/bugs/download.php?file_id=50447)

This issue was migrated from https://savannah.nongnu.org/bugs/?54977

avrs-admin, 2022-01-31:
Gregor Riepl <onitake>
Sat 12 Dec 2020 06:14:08 PM CET

The data sheet explicitly mentions PCMSK0 for the tiny2313A and tiny4313. There is no mention of PCMSK.

Adding PCMSK0 while leaving PCMSK in place for compatibility with older 2313 code should not cause any issues.

In the hopes of getting more attention, I'm attaching a patch to add the missing macro.

(**[file #50447](https://savannah.nongnu.org/support/download.php?file_id=50447)**)

sprintersb, 2024-03-14:
Applied as https://github.com/avrdudes/avr-libc/commit/74a7d34ef2ecba31180d7c172b7fa026e1981de7.

closed 2024-03-14 by sprintersb