avrdudes/avr-libc #671: [bug #57114] Do not #define abs / labs

avrs-admin, 2022-01-31:
Fri 25 Oct 2019 10:42:00 AM CEST

#define'ing abs is evil, in particular together with C++ when there are methods called abs or there is abs polymorphism.

Same for labs.

The patch provides proper functions (kind of) without introducing overhead.

```
* include/stdlib.h (__ATTR_ALWAYS_INLINE__): New macro.
(abs): Provide in such a way that we get a prototype.
(labs): Same.

```

[file #47738: stdlib-abs.diff](https://savannah.nongnu.org/bugs/download.php?file_id=47738)

This issue was migrated from https://savannah.nongnu.org/bugs/?57114

sprintersb, 2022-09-20:
This patch / issue is no more needed and should not be applied.  The respective change has been made by the math.h rework as of

https://github.com/avrdudes/avr-libc/commit/556b4a8b3fee2be347c9f40f05a9cb2d891ea78b#diff-23ba326468a35d50afdafb6950908d5618bab93d3eabfea4ce89e63462feb0b1

sprintersb, 2024-01-31:
Fixed with 556b4a8b3fee2be347c9f40f05a9cb2d891ea78b

closed 2024-01-31 by sprintersb