avrdudes/avr-libc #675: [bug #58005] Support for new ATtiny/ATmega devices

avrs-admin, 2022-01-31:
Tue 17 Mar 2020 04:14:31 PM CET

The devices I'd like to see supported are (at least):

- ATtiny102/104
- ATtiny20x/40x/80x/160x
- ATtiny21x/41x/81x/161x/321x
- ATmega320x/480x

I've had a look at both the repository and the bugtracker and
couldn't see any work-in-progress to add support for this set
of new devices. If there is some effort I missed, I'd be very
grateful for a link.

I'm aware that Microchip/Atmel offer support for these devices
through their download packs. However, I'd much prefer support
to be in avr-libc itself, as using these packs is a major pain
(unless, I reckon, you're using Atmel Studio, which I don't).

It seems to me that Atmel have already provided code to the
avr-libc project in the past, so I'd hope that this would be
no different in case of these devices. I'm probably just being
impatient here given that the header files in the download
packs are dating from this year. :-)

I'd be more than happy to help if there's anything I can do.

Thanks,
Marcus

This issue was migrated from https://savannah.nongnu.org/bugs/?58005

avrs-admin, 2022-01-31:
Yves Delley <burnpanck>
Thu 14 May 2020 07:04:46 AM CEST

See #9543
 [https://savannah.nongnu.org/patch/?9543](https://savannah.nongnu.orghttps://savannah.nongnu.org/patch/?9543)

sprintersb, 2024-03-15:
* ATtiny102/104: Added in e3e3768285b3c0e3b9888fb931b3494b956f7398.

* ATtiny20x/40x/80x/160x. Added in 960a359d854001399b8fb445a9ad0ce3ea80f4d0.

* ATtiny21x/41x/81x/161x/321x: Same.
 
* ATmega320x/480x: Same.