avrdudes/avr-libc #677: [bug #58711] correcting errors in documentation inquiry

avrs-admin, 2022-01-31:
Sun 05 Jul 2020 11:41:26 PM CEST

- as it relates to .dox filesIn the doxy source for the manual, how should spelling/grammar related corrections be handled? The rate at which I come across areas in the documentation in need of corrections is rather low.  I also manage to forget about them shortly afterwards and don't presently write them down (but could use highlighting in the PDF I just realized). I did check if the online manual had the same error as well, thus this submission.

I realize it's tremendously less critical than the actual avr library so is this something even worth tracking externally or is it something that the crew is likely to find eventually and correct. I am cloning the SVN since even attaching the file to diff would take a lot less time than describing where attention would be needed.

Thanks!

This issue was migrated from https://savannah.nongnu.org/bugs/?58711

avrs-admin, 2022-01-31:
Joerg Wunsch <joerg_wunsch>
Mon 06 Jul 2020 07:40:08 AM CEST

It is completely fine to issue tickets for such errors.

avrs-admin, 2022-01-31:
Steven Kirkland <lemonforest>
Mon 06 Jul 2020 08:46:40 PM CEST

Ok, thanks. I'll make corrections as I find them then, as I get time, rebase and go back through the entire chapter before opening a ticket unless it'd be better to just keep it all in here.

Thanks!

sprintersb, 2024-07-31:
Question has been answered.

closed 2024-07-31 by sprintersb