avrdudes/avr-libc #678: [bug #58862] avr/wdt.h LLVM11 error for atmega328p

avrs-admin, 2022-01-31:
Fri 31 Jul 2020 01:32:44 PM CEST

Error message:

avr\include\avr/wdt.h:419:12: error: value '64' out of range for constraint 'I'
: "I" (_SFR_IO_ADDR(_WD_CONTROL_REG)),
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
avr\include\avr/sfr_defs.h:183:27: note: expanded from macro '_SFR_IO_ADDR'
#define _SFR_IO_ADDR(sfr) (_SFR_MEM_ADDR(sfr) - __SFR_OFFSET)
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

-------------------
I guess that wdt_enable()/wdt_disable() should have conditional preprocessor code instead of conditional inline.

This issue was migrated from https://savannah.nongnu.org/bugs/?58862

KOLANICH, 2022-06-08:
I guess that the quickest fix is to use `& 0b111111` for "runtime dispatch" [currently present](https://github.com/avrdudes/avr-libc/blob/960a359d854001399b8fb445a9ad0ce3ea80f4d0/include/avr/wdt.h#L450-L532) to calm down the compiler. It seems that that branch is not expected to be taken anyway on those MCUs.

sprintersb, 2024-04-12:
The problem is in clang. Even with `__builtin_constant_p` it compiles a block with a non-matching constraint, but it's the very purpose of `__builtin_constant_p` to skip compilation of `if (0)` blocks. Test case:

```c++
#include <avr/io.h>

#define SFR (*(volatile char*) (0x40 + __SFR_OFFSET))

void func (void)
{
    if (__builtin_constant_p (_SFR_IO_REG_P (SFR))
        && _SFR_IO_REG_P (SFR))
    {
        __asm volatile ("in __tmp_reg__,%0"
                        :: "I" (_SFR_IO_ADDR (SFR)));
    }
}
```
Notice that `_SFR_IO_REG_P (SFR)` is a compile-time constant that evaluates to `0`.  Even simpler test case:
```c++
void func (void)
{
    if (0)
        __asm volatile ("in __tmp_reg__,%0" :: "I" (64));
}
```
Closed. This is a clang/llvm issue.

closed 2024-04-12 by sprintersb

UltraBlackLinux, 2024-07-02:
@sprintersb I've also just run into this issue, but I don't really understand what's going on with this problem at all. 
Would you mind creating an issue over at [llvm](https://github.com/llvm/llvm-project/issues/new) regarding this problem? I think I'd grossly misrepresent this problem if I opened an issue there myself.

I myself kinda depend on clang to get language servers working, and tooling that does exist for arduino often doesn't properly interface with the software I'm using. `clang` and it's `compile_commands.json` seems to be the only good way to get reliable code completion and syntax highlighting within my editor.